AllStarLink/app_rpt #587: Asterisk Looking for Audio Files That Don't Exist and Throwing Warnings in Log

ve7fet, 2025-03-24:
Fresh base install of ASL3 on Debian 12:

 OS            : Debian GNU/Linux 12 (bookworm)
 OS Kernel     : 6.1.0-32-amd64
 
 Asterisk      : 22.2.0+asl3-3.3.0-6.deb12
 ASL [app_rpt] : 3.3.0
 
 Installed ASL packages :
 Package                         Version
 ==============================  ==============================
    asl3                            3.7.1-1.deb
    asl3-asterisk                   2:22.2.0+asl3-3.3.0-6.deb12
    asl3-asterisk-config            2:22.2.0+asl3-3.3.0-6.deb12
    asl3-asterisk-modules           2:22.2.0+asl3-3.3.0-6.deb12
    asl3-menu                       1.12-1.deb12
    dahdi                           1:3.1.0-2

Open the console (or tail the messages.log) and connect to another node.

Asterisk throws a fit and claims it can't find the audio files:

```
asl3*CLI> rpt fun 521802 *3435455
    -- Call accepted by 44.32.40.212:4569 (format ulaw)
    -- Format for call is (ulaw)
    -- Hungup 'IAX2/44.32.40.212:4569-352'
    -- Hungup 'DAHDI/pseudo-1227379051'
    -- Hungup 'DAHDI/pseudo-1366019158'
[2025-03-24 16:16:50.445] WARNING[6783]: file.c:825 ast_openstream_full: File custom/rpt/node does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'rpt/node.gsm' (language 'en')
[2025-03-24 16:16:51.086] WARNING[6783]: file.c:825 ast_openstream_full: File custom/digits/4 does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'digits/4.ulaw' (language 'en')
[2025-03-24 16:16:51.888] WARNING[6783]: file.c:825 ast_openstream_full: File custom/digits/3 does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'digits/3.ulaw' (language 'en')
[2025-03-24 16:16:52.727] WARNING[6783]: file.c:825 ast_openstream_full: File custom/digits/5 does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'digits/5.ulaw' (language 'en')
[2025-03-24 16:16:53.549] WARNING[6783]: file.c:825 ast_openstream_full: File custom/digits/4 does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'digits/4.ulaw' (language 'en')
[2025-03-24 16:16:54.352] WARNING[6783]: file.c:825 ast_openstream_full: File custom/digits/5 does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'digits/5.ulaw' (language 'en')
[2025-03-24 16:16:55.174] WARNING[6783]: file.c:825 ast_openstream_full: File custom/digits/5 does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'digits/5.ulaw' (language 'en')
[2025-03-24 16:16:55.995] WARNING[6783]: file.c:825 ast_openstream_full: File custom/rpt/connection_failed does not exist in any format
    -- <DAHDI/pseudo-53154922> Playing 'rpt/connection_failed.gsm' (language 'en')
    -- Hungup 'DAHDI/pseudo-53154922'
    -- Hungup 'DAHDI/pseudo-1880530281'
```

It actually does play the audio files, it appears to find them in `/usr/share/asterisk/sounds/en`, but then it also appears to want to source them a second time from (presumably) `/var/lib/asterisk/sounds/custom`.

As a result, it fills the console (and the log file) with the above warnings about not being able to find the sounds.

Allan-N, 2025-03-24:
A fix for this is coming with https://github.com/AllStarLink/asl3-asterisk/pull/46

closed 2025-03-24 by Allan-N

Allan-N, 2025-03-24:
Also, as a workaround, you can update your `/etc/asterisk/asterisk.conf` and set `sounds_search_custom_dir = no`

ve7fet, 2025-03-24:
Thanks Allan, looking forward to the update, glad it has been addressed.