AllStarLink/app_rpt #599: archivedir filename with higher time accuracy

kenalker, 2025-03-31:
The comments in rpt.conf for the "archivedir" functionality state:

> ; The following "archivedir" line can be used to enable a simple log and
> ; audio recorder of the activity on a node. When enabled, a series of
> ; recordings, one for each active COR on the node. The file(s) will be
> ; named with the date and time down to the 1/100th of a second. This
> ; logging can be useful in debugging, policing, or other creative things.

I wrote a rather complex script this weekend that concatenates archivedir generated WAV files over the course of a pre-determined timeframe and, in the process, recreates and injects silence between each WAV file to simulate the original amount of silence that would have existed when the WAV files were recorded.  In order to calculate original gaps between WAV files, I was using the UNIX birthdate of the WAV files (via stat command) and duration of WAV files (via ffprobe).  However, I found that the UNIX birthdate changes when files are copied.  Thus, I would instead like to rely on the timestamp that is found in the filename.  I need better than the 1-second precision that is currently offered by the filenaming convention (ie. 20250330213001.WAV).  The above comment states that the filenames should have 2 more digits in them (1/100th seconds), yet they do not.  I am in hopes that someone will re-write the code so that the filename is more time-precise, as the above comment states.

I'm guessing that apt_rpt.c v3.4.1 line 6969 will need to use something like gettimeofday instead of locatime (but I'm no expert so just a guess).

Thank you!

Allan-N, 2025-03-31:
>However, I found that the UNIX birthdate changes when files are copied.

I don't know how you are copying the files but if you are using shell commands then try `cp -p <old> <new>`.

>The above comment states that the filenames should have 2 more digits in them (1/100th seconds), yet they do not. I am in hopes that someone will re-write the code so that the filename is more time-precise, as the above comment states.

Here's the gotcha.  The `rpt.conf` comments say one thing the code and the time-stamps used in the filename(s) and in the .txt log files are doing something different.  In order to not break any existing programs/scripts we need to be careful about any changes (other than documentation).  I'm thinking that any timestamp updates will need to be  gated by a new configuration setting ... something like `archivetime = ms`.

kenalker, 2025-03-31:
I am using `cp -p` but that only preserves the modify date, not the birthdate (I know - weird - who would have thought?).  But I'd still like this feature because someone else might move the files w/o thinking to preserve the dates (including me, in the future), or they may transfer to another operating system which may munge the dates, etc.  The ultimate way to preserve is to have it as the filename.

I agree with the configuration setting idea.  What would be REALLY nice is if there was a way to dictate the filename convention via a user settable variable.  I think this is done in Asterisk via the MONITOR_FILENAME variable, if I'm not mistaken.  That said, since Allstar is one step removed from the monitor() function, I'm guessing this would be a lot of code and time consuming, especially to incorporate various flavors of timestamping, so I didn't go that far in my request.

Allan-N, 2025-03-31:
It looks like `ast_localtime()` and `ast_strftime()` will help ... with the latter providing support for including "%q" in the format string if we want to include milliseconds.

closed 2025-04-21 by Allan-N

kenalker, 2025-04-22:
Thank you Allan!  I'm looking forward to trying this and then completing my scripts (and releasing to the community).  I'll wait for the next Debian package release (assuming that is the only way to easily get new code other than re-compiling myself).  Again, thank you for your efforts and time commitment.

kenalker, 2025-07-27:
I compiled the code and tested and this works - thank you!

BTW, If I enter a "0" in front of the q in the date/time string then I get an output file named like this: "20250727002042%0q.WAV".  I am not sure if you want to test for "0q" and force the filename to be the same as having no "q" in the string at all, or just ignore, but figured I'd point this out.

Allan-N, 2025-07-27:
>BTW, If I enter a "0" in front of the q in the date/time string then I get an output file named like this: "20250727002042%0q.WAV". I am not sure if you want to test for "0q" and force the filename to be the same as having no "q" in the string at all, or just ignore, but figured I'd point this out.

Honestly, I would like to stay away from manipulating the format string.  Hopefully, the `rpt.conf` comment about the format string and example will be sufficient.
```
                                       If desired, you can specify any
; date format supported by the C function strftime(3) API.  For more
; precision, you can also include the '%[n]q' format to add fractions
; of a second, with leading zeros.
```
```
;archivedatefmt = %Y%m%d%H%M%S%2q         ; date/time (time to 1/100th secs)
```