AllStarLink/app_rpt #600: archivedir recording (mostly) silence to WAV file when duplex=3

kenalker, 2025-03-31:
**Describe the bug**
I am using the “archivedir” function to record repeater audio.  I have a single repeater with an RTCM attached to the ASL3 computer.  Audio is recorded to a WAV file (date_and_time_stamp.WAV as filename) each time someone keys up until they unkey (one file per key-up), as expected. 
 The files contain audio when duplex=2 in rpt.conf.  However, when duplex=3 in rpt.conf the files contain no received audio, but they do contain the courtesy tone and ID in CW (if present) at the end of the recording.  WAV files are of the appropriate size to contain audio and `ffprobe -v error -show_format 20250331003700.WAV` reports a file of 8-second duration.

**To Reproduce**
Steps to reproduce the behavior:
1) Add archivedir=/var/spool/asterisk/monitor to your node stanza in rpt.conf.
2) Key up your repeater and speak.
3) Play back the WAV file created in the /var/spool/asterisk/monitor/##### directory (where ##### is your actual node number)

**Expected behavior**
I expect to have a WAV (*.wav) file recorded that  has spoken audio from the person who keyed up the repeater and spoke, however, with duplex=3 in rpt.conf I get only silence followed by a courtesy tone (and sometimes the repeater ID in CW) at the end.

**Software versions (listed in asl-menu, option 4)**
```
OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-32-amd64
Asterisk      : 22.2.0+asl3-3.3.0-6.deb12
ASL [app_rpt] : 3.3.0
Installed ASL packages :
Package                         Version
==============================  ==============================
allmon3                         1.4.2-1.deb12
asl3                            3.7.1-1.deb
asl3-asterisk                   2:22.2.0+asl3-3.3.0-6.deb12
asl3-asterisk-config            2:22.2.0+asl3-3.3.0-6.deb12
asl3-asterisk-modules           2:22.2.0+asl3-3.3.0-6.deb12
asl3-menu                       1.12-1.deb12
dahdi                           1:3.1.0-2
dahdi-dkms                      1:3.4.0-6+asl
dahdi-linux                     1:3.4.0-6+asl
```

**Have you run a software update and rebooted?**
Yes.

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Desktop - Intel(R) Core(TM) i5-2300 CPU @ 2.80GHz

mkmer, 2025-03-31:
This is caused here:
https://github.com/AllStarLink/app_rpt/blame/e554ae424e653f28ce468571faf05ab1e953f318/apps/app_rpt.c#L2734-L2738

We put duplex mode 3 in the same conference configuration as duplex mode 1.  I suspect this is to prevent the RX audio from coming out the speaker on the node while still allowing audio on the links to come through.  As a result the monitor channel is working fine, but there is no audio on the TX conference to record.  The telemetry does come in via the TX conference, so we record that. 

I don't see an "easy" fix for this at the moment.

kenalker, 2025-04-22:
Thank you for your time and effort on this!  Is there a way for me to test this without waiting for the next Debian package, other than re-compiling entirely from scratch?

mkmer, 2025-04-22:
Compiling is the only way right now.  We are in a PR backlog at the moment, and as we merge there WILL be some bugs so it's probably not the best time to test (but we do need early feedback too).

kenalker, 2025-07-01:
It appears that this fix got merged in so I upgraded to the 22.4.1+asl3-3.5.2-1.deb12 build today.  However, the bug is still present.  Maybe I'm misinterpreting the build contents; did I jump the gun on my upgrade?

Allan-N, 2025-07-01:
FYI : we had a fix for `duplex = 3` archive audio but, unfortunately, that changeintroduced a [regression](https://github.com/AllStarLink/app_rpt/issues/703) and needed to be backed out.

We have another potential fix, trying to address the issue in a different way, that we hope to include in our next release.

kenalker, 2025-07-02:
Is it in the new release?  I did apt-update and apt-upgrade and now on 22.4.1+asl3-3.5.4-1.deb12 but I'm still getting silent recordings (other than telemetry).

Allan-N, 2025-07-03:
>Is it in the new release?

No, [#708](https://github.com/AllStarLink/app_rpt/pull/708) is NOT in the 3.5.x release.  After the proposed changes have been reviewed and tested we'll get it merged into the main source code branch.  Hopefully, we'll get this in our next set of packages.

Please let us know if you are willing to "test" out the changes out ahead of time.

kenalker, 2025-07-03:
Happy to test!  As I've gotten older I've grown to rely on the path of least resistance (apt-get update) and have somewhat lost my knowledge on compiling things (even though that's the only way I did anything for many years, but that was 30 years ago, and things have gotten more complex since then).  I would just need a bit of a refresher on what I need to do to compile it, assuming that's what I'd have to do since not in the latest release.

I've been pouring over the releases of the various ASL3 packages for weeks now, and, admittedly, I've not been able to make sense out of the versioning/dependencies in GitHub so I've been hesitant to compile, but certainly willing (and able, barring some (re-)learning).

closed 2025-07-20 by Allan-N

kenalker, 2025-07-27:
I compiled and tested duplex=0,1,3,4 and all seem to work!  I just did one recording in each mode using FM and Allstar as my source.  I didn't do anything extensive (like listening to conversations for any other types of anomalies as someone else found in a previous release that had to be unwound).

mkmer, 2025-07-27:
Let's hope I didn't break anything else :)

commit b914f93d (2025-03-31): app_rpt: Update duplex mode 3 to pass audio to audioarchive.

Mode 3 is intended to be a full duplex with local audio muted.  This WAS accomplished via conference settings but that prevented the local audio from reaching the monostream channel.
Converting the conference settings to match modes 2 and 4 (full duplex) and simply muting the frames headed to the txchannel allows for monostream to catch the audio and prevent the audio from feeding back on the local speaker.
Resolves issue archivedir recording (mostly) silence to WAV file when duplex=3
Resolves #600
commit 38b0d7e7 (2025-03-31): app_rpt: Update duplex mode 3 to pass audio to audioarchive.

Mode 3 is intended to be a full duplex with local audio muted.  This WAS accomplished via conference settings but that prevented the local audio from reaching the monostream channel.
Converting the conference settings to match modes 2 and 4 (full duplex) and simply muting the frames headed to the txchannel allows for monostream to catch the audio and prevent the audio from feeding back on the local speaker.
Resolves issue archivedir recording (mostly) silence to WAV file when duplex=3
Resolves #600
commit 6d934b0d (2025-04-01): app_rpt: Update duplex mode 3 to pass audio to audioarchive.

Mode 3 is intended to be a full duplex with local audio muted.  This WAS accomplished via conference settings but that prevented the local audio from reaching the monstream channel.
Converting the conference settings to match modes 2 and 4 (full duplex) and simply muting the frames headed to the txchannel allows for monstream to catch the audio and prevent the audio from feeding back on the local speaker.
Resolves issue archivedir recording (mostly) silence to WAV file when duplex=3
Resolves #600
commit e5a4fc03 (2025-04-11): app_rpt: Update duplex mode 3 to pass audio to audioarchive.

Mode 3 is intended to be a full duplex with local audio muted.  This WAS accomplished via conference settings but that prevented the local audio from reaching the monstream channel.
Converting the conference settings to match modes 2 and 4 (full duplex) and simply muting the frames headed to the txchannel allows for monstream to catch the audio and prevent the audio from feeding back on the local speaker.
Resolves issue archivedir recording (mostly) silence to WAV file when duplex=3
Resolves #600