AllStarLink/app_rpt #614: Echolink not "haning up" channels after call

mkmer, 2025-04-04:
Not sure how I missed this in testing, but we are not hanging up the channels after a call is completed.

After moving `softhangup()` out of `find_delete()` there are 2 locations where we need to hangup, but no longer have a channel reference to hangup.

https://github.com/AllStarLink/app_rpt/blob/73f70323c84692f4561680b5a7aeb28bd8451180/channels/chan_echolink.c#L2175-L2178

and here:

https://github.com/AllStarLink/app_rpt/blob/73f70323c84692f4561680b5a7aeb28bd8451180/channels/chan_echolink.c#L3686-L3689

We have a channel here, so probably just need a `softhangup()`

https://github.com/AllStarLink/app_rpt/blob/73f70323c84692f4561680b5a7aeb28bd8451180/channels/chan_echolink.c#L2373-L2375

InterLinked1, 2025-04-04:
We should create tests that fire up a channel of each type. That most likely would've caught the issue since Asterisk would not have been able to shut down cleanly.

mkmer, 2025-04-04:
Sounds "easy", but in order to actually test an echolink channel, we need to emulate the socket side with all the the commands that echolink sends and expects to receive.  
This certainly should be done, but it's far from trivial.

mkmer, 2025-04-04:
This is my read of what is going on (many fine details missing):
It looks like the `el_reader` thread listens for new udp type messages, translates those messages (by call sign, IP, db) into a channel.  Each node talking on the UDP port get's it's own asterisk channel and we translate voice packets into queues with are then read/write in the `xwrite` function.
The audio then goes into the "normal" RPT_CONF which each channel is connected to.  So the audio comes in Echolink, goes into the conference, comes back out the conference on a different channel, back to Echolink, back out to the correct IP address via UDP.

Before any of that can happen, we do a bunch of "checks" to see if the callsign is in a database or whatever to allow the connection to come up.

PR #615 (2025-04-07): chan_echolink: restore removed hangup conditions
Relocaiton of `softhangup()` outside of the `find_delete()` function left channels hanging.
The channel's private data is available and sticks with a channel masquerade.  Simply add a flag in the private data allows us to hangup on the next channel read/write.
Fixes #614