AllStarLink/app_rpt #619: xpmr.c does not handle `ast_calloc()` failures

mkmer, 2025-04-09:
One example (`->x` is `ast_calloc()`'d in multiple places  :
https://github.com/AllStarLink/app_rpt/blob/7695c7e9bfe6bd6bc7674e960d60dddeb8a0be4b/channels/xpmr/xpmr.c#L2508-L2509

Not sure the "right thing" to do at the moment.

Allan-N, 2025-04-09:
IMHO, any piece of software that encounters a memory allocation failure has much bigger issues.  Yes, in the days before virtual memory one needed to be much more concerned with memory usage.  Today, if memory is not available then you clearly asked for way too much or you've completely filled up your swap space.

For the case you called out, you don't need the printf() 'cause that's handled by ast_calloc().  The only question is whether you want/need to deal with the failed allocation.

mkmer, 2025-04-09:
> IMHO, any piece of software that encounters a memory allocation failure has much bigger issues. Yes, in the days before virtual memory one needed to be much more concerned with memory usage. Today, [...]

Agreed, but if it fails, a crash is not ideal.  At the moment, it's not clear to me what to do if it fails, so for now, it does what it always did and keeps moving on, probably resulting in crash due to a NULL pointer.

I've created and issue so we don't "forget" we need to do something at some point.

InterLinked1, 2025-04-09:
> The only question is whether you want/need to deal with the failed allocation.

It's not really a question. In Asterisk, this is mandatory. We do need to address this at some point.

Things can fail even in production and crashing is not an acceptable fallback. For example, there was a period of time where `fork()` would frequently fail on my production system due to memory fragmentation.

Allan-N, 2025-04-09:
>For example, there was a period of time where fork() would frequently fail on my production system due to memory fragmentation.

and this was just an issue just for the asterisk process?  or did the memory fragmentation issue also affect many other processes on your system? 

Note: I'm not saying that we shouldn't check for / handle errors.  But, at the same time, if you are adding lots and lots of error handling code that will likely never be used then you are adding to the memory footprint of the process, something that increases the time to load the process, increases the need for VM, can have an impact on swap, etc.

InterLinked1, 2025-04-09:
> > For example, there was a period of time where fork() would frequently fail on my production system due to memory fragmentation.
> 
> and this was just an issue just for the asterisk process? or did the memory fragmentation issue also affect many other processes on your system?

Just Asterisk. That's where the memory fragmentation was.

> Note: I'm not saying that we shouldn't check for / handle errors. But, at the same time, if you are adding lots and lots of error handling code that will likely never be used then you are adding to [...]

That's a fair point. But for a long running daemon like Asterisk, which is meant to be production grade, the decision has already been made by the project that that is the way to go. No code is accepted that does not handle possible failures. In the case of xpmr.c compared to the rest of Asterisk, it's really peanuts.

Allan-N, 2025-04-09:
>Just Asterisk. That's where the memory fragmentation was.

So, the fork() failures were a good thing because they highlighted the memory fragmentation problem :)

davidgsd, 2025-10-03:
From what I can tell, this "x" member of pSps is never actually used, ie. written to or read from - other than the calloc.  Thus looks like all the following lines can all be safely deleted:

```
allscan@node571570:~/app_rpt$ gc '->x '
./channels/xpmr/xpmr.c:398: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:399: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:410: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:411: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:435: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:436: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:447: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:448: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2034: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2035: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2063: pSps->x = ast_calloc(pSps->nx, pSps->size_coef);
./channels/xpmr/xpmr.c:2064: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2098: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2099: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2140: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2141: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2168: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2169: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2266: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2267: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2296: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2297: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2407: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2408: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2462: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2463: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2596: if (pSps->x != NULL)
./channels/xpmr/xpmr.c:2861: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2862: if (pSps->x == NULL) {
./channels/xpmr/xpmr.c:2873: pSps->x = ast_calloc(pSps->nx, pSps->size_x);
./channels/xpmr/xpmr.c:2874: if (pSps->x == NULL) {
```

I will probably be doing a PR soon relating to USBRadio so could also delete these lines if someone would like to take a 2nd look at them and confirm I'm not missing something and if they maybe actually are being used somewhere somehow.