AllStarLink/app_rpt #624: ASL3 Debian12 glitches in USB Rx audio

davidgsd, 2025-04-12:
I have a node that had and still has some intermittent glitches in the USB rx audio (from URI ADC). This node had not been updated in a few months so I updated it to latest ASL3.3 and the issue was still there. I then tried updating to latest devel and that doesn't fix it either. sudo asterisk -V shows: Asterisk 22.2.0+asl3-3.4.2-1.deb12

I have another node with the same URI and everything is fine on that, it has Asterisk 20.11.0+asl3-3.2.0-2.deb12.

Here is what the glitches look like:
glitches
[glitches1920×1131 175 KB](https://community.allstarlink.org/uploads/default/original/2X/5/5c3afde68a2ab8affb409f7bbbd89aba756ff6d9.jpeg)

(from parrot wav file recorded by the node in /tmp/)
ie. there are anywhere from 5 ~15 zero's in the PCM data, but only happens intermittently, maybe once every 10 to 20 seconds of audio, with maybe 8 or 9 out of 10 20mS USB frames then having a sequence of zero sample values (presumably at frame boundaries), resulting in some choppiness/distortion in the audio.

Please LMK of any suggestions on how to further debug this issue.

**To Reproduce**
Set up a node and do a parrot test and listen for distortion / "choppiness" in the played back audio. My node has duplex=3 and is using SimpleUSB channel driver.

**Have you run a software update and rebooted?**
 - YES

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Dell Wyse 3040 MiniPC

davidgsd, 2025-04-15:
Did some further testing and it appears the bug is in fact on USB Tx audio and not in Rx audio.  It happens rarely enough that it just sounds like some occasional static so at first it did not seem it was happening in incoming IAX audio but in fact is.  (Recording USB Rx audio with susb tune rxcap seems to work fine.) 

I do not see any "stuck USB" or any other warning/error messages when the glitches occur.

I attached 2 files with the glitches:
[wavs.zip](https://github.com/user-attachments/files/19753136/wavs.zip)

In one file I did a parrot test and right in the very beginning of the courtesy tone there were 2 glitches 20mS apart:

[image]

Since this glitch was in the courtesy tone itself that clarifies the issue was not in incoming USB or IAX audio ie. is definitely happening somewhere between Asterisk and the USB driver.

Then in the other file I was recording incoming IAX audio and in this example there are 2 full 20mS frames that got zeroed followed by a shorter glitch 20mS later.

mkmer, 2025-04-15:
Ok - TX side is something we've been having trouble with.  I added this code recently after discovering we can't keep the USB buffer full:
https://github.com/AllStarLink/app_rpt/blob/7695c7e9bfe6bd6bc7674e960d60dddeb8a0be4b/channels/chan_simpleusb.c#L2075-L2080
On my systems, it only triggers 1 time at the beginning of a transmission, but when I was measuring the update times it appeared the update was always + on the 20ms never (-) - 20.01, 20.02, etc.  If we fall behind enough to under run the buffer, it would trigger again.  

You set `core set debug 7 chan_simpleusb` and we can see if this is actually triggering in the middle of a transmission.

I suspect we need to figure out how to write on average 20ms 100% of the time to resolve the issue.

davidgsd, 2025-04-15:
Setting debug to 7 isn't very helpful because a 'blocks used...' message is printed 100 times per second and thus the terminal scroll buffer overflows before much other info is shown. So I can't say for sure if the 'A null frame has been added' message appears every time there is a glitch, but I do see the message sometimes. Maybe the 'blocks used' message is not needed or could be reduced in frequency or only printed if there was actually something unexpected that happened?

[image]

mkmer, 2025-04-15:
Well - it is showing that the buffer is staying "full" -> it would be helpful if you captured an event and we could prove/disprove this code is causing the problem.

The strange part about THIS code is that  the `simpleusb_write()` function is accepting frames into a buffer which is then read out in the `simpleusb_read()` function.  I'm suspicious of the delay this creates and the possibility of doing a read before a write, which would result in an empty tx buffer for one of the cycles.  
All in all, the USB device's "output" is driving the timing via a file device trigger (which I believe is a polled event rather than interrupt driven - more research required).
I'm not sure WHY we are not driving writes directly in the write function (as one would assume it should be).  I'm planing on refactoring it when I get a bit of time and see what blows up.  Always a bit of fun...

davidgsd, 2025-04-15:
I know I have heard these little glitches before on other nodes in the past, I think for quite some time, but they were minor enough that I didn't look further into it. I will do some other testing on other nodes, but just from the fact that we have a USB interface outputting 48Ksps at exact 20mS frames (96 samples/frame, upsampled from 8Ksps 16 samples/frame) - yet the channel driver is doing writes closer to 22mS apart, there is something seriously wrong here and this seems like a major bug. Does Asterisk maybe have a reference implementation for how USB writes should be done in a channel driver?

wd5m, 2025-04-16:
I may be seeing this issue on my ASL3 VPS Hub too. It has no audio devices, other than DAHDI, to provide a network HUB for Allstar. I've configured node 614320 to demonstrate what this sounds like. When you connect to 614320, it will auto connect to the ARRLNEWS feed to provide audio. You'll hear the audio dropouts every few seconds as the news plays back. 

This ASL3 VPS HUB is running the latest release (NOT DEV/TEST) of ASL3 on Debian 12 (6.1.0-33-amd64) in a Vultr VPS.

David, WD5M

davidgsd, 2025-04-16:
I have not confirmed this 100% but I suspect that what is happening is that this may only happen on nodes that had ASL originally set up more than 2-3 months ago eg. before the more severe audio choppiness issue happened when the kernel updated from xx.28 to xx.30 and then subsequent updates were rolled out to fix that. It appears that if a new node is set up with a fresh install of Deb 12 and latest ASL that everything works fine. I just built a new node today with a Beelink T5 miniPC and am not hearing any glitches from it, and I do not believe I have heard any glitches from a couple other new nodes I built in the past couple weeks. So my suspicion is that when updating an existing node there is something not properly updated, that then causes this issue. For the time being I guess the recommendation should be to not update older nodes, but that's not a real solution as it would force a lot of people to have to do a complete reinstall if they ever want to update.

Allan-N, 2025-04-16:
So, you have 1 x86_64/amd64 system with the glitches and one without?  If my memory serves me, the only change we were fussing with when the kernel-related choppiness issue arose was to add the `lowlatency` option to the `snd_usb_audio` module.  It would be good to compare your two systems to see which file (or files) exist that are setting the option ...  and to what value.

On my RPi[4] I have `/etc/modprobe.d/asl3-snd-usb-audio.conf` with :
```
options snd_usb_audio lowlatency=0
```

Do you have the same on both of your systems?  Do you have any other module .conf files that are adjusting the `lowlatency` setting?  What does `modprobe -c | grep lowlatency` report?

Allan-N, 2025-04-16:
Assuming that both of your systems (the one with the choppiness and the one without) both report the same lowlatency value ... and if your thought that nodes that were installed months back and have been updated/upgraded are behaving differently from fresh installs is true ... then the question (we all) need to ask is what [Debian?] configuration files are not being updated.

jxmx, 2025-04-16:
Are both nodes running the same kernel - `uname -a`? Are both nodes running the same version of DAHDI - `dpkg -l | grep dahdi`? I have a number of "old" systems that have been continuous upgraded for over a year with ASL3 and experience no audio chop problems.

jxmx, 2025-04-16:
Also, how are you recording the audio precisely? Some ways of getting the audio would have nothing to do with the USB channel driver.

mkmer, 2025-04-17:
I'm also running the Dell 3040 in two instances, I haven't been able to "find' the same behavior.  One of my nodes are running a bit older kernel (not sure why - apt update should have me on the latest), the other a bit newer.
`Linux wyse 6.1.0-30-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.124-1 (2025-01-12) x86_64 GNU/Linux`
`Linux wyse2 6.1.0-33-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.133-1 (2025-04-10) x86_64 GNU/Linux`

If you ARE seeing these audio drops in the `tx_cap_in.pcm`, then it's coming to the `chan_simpleusb` driver that way - so something in the path (IAX2, ulaw driver, slin driver, source audio...)  is doing the deed.

I "thought" you had said these are < 20ms drops (Is that still correct?).  Any "muting" done in asterisk is an entire frame - so there really shouldn't be anything less than 20ms or 20ms increments if it was caused by an internal "mute".

jxmx, 2025-04-17:
@davidgsd can you please capture audio from outstreamcmd or the archive function based on network-based (non RF/susb) traffic? It would be very helpful to know if the problem is occurring in simpleusb or app_rpt.

davidgsd, 2025-04-18:
Testing done:
1. Everything is working great on a Beelink T5 that has not been updated since Nov.:
Linux t5 6.1.0-28-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.119-1 (2024-11-22) x86_64 GNU/Linux

2. On one of the 3040's I used 'outstreamcmd = /bin/sh,-c,/usr/bin/lame -r -m m -s 8 --bitwidth 16 - /tmp/txcap.wav' in rpt.conf, and 'susb tune txcap' and confirmed that glitches are present in both files. See below attached. The compression being done by lame is very high (no idea why - I'm writing to a wav) so it's not as clear there but if you look in a wave editor there is a stream of 8 zero samples at 6.293s into tx_cap_in.wav, and a corresponding glitch at the same place in txcap.wav (not zeroes though because was coded/decoded apparently to/from mp3 ie. FFT'd.) So this apparently confirms the issue is not in SimpleUSB.
[txcaps.zip](https://github.com/user-attachments/files/19809212/txcaps.zip)

jxmx, 2025-04-18:
And to be 100% clear, the audio in question was generated by `rpt localplay` or a network-based source?

davidgsd, 2025-04-18:
Yes the audio is local, a DTMF command that calls the Playback function.

davidgsd, 2025-04-22:
Following up on this bug. Can someone unmark it from "wait-on-user"?  From testing done as described above this is a major issue happening on both old and new installs on multiple nodes. The resultant audio glitches are "relatively" minor, just a bit of distortion every 5-10 seconds but that is definitely not acceptable in use cases such as repeaters or in commercial node products. The glitches seem to happen only on some mini PCs eg. Dell 3040 w/1.44GHz Atom CPU and 2GB RAM but not on others eg. Beelink T5 with higher spec Celeron CPU and 4GB RAM. But this issue definitely was not happening on ASL3 from a few months ago (w/Linux 6.1.0-28-amd64). Thanks

davidgsd, 2025-04-24:
Set up another new node today, Dell 3040, installed Deb 12 (net install 'CD'), then ASL3, Allmon3 & AllScan. When I first did a parrot test the issue was not there and I thought things were looking up but then after a reboot I noticed the glitches again in the telemetry and IP announce. Quite a few glitches just in the first 8 seconds, including one in the courtesy tone and a relatively long 12mS glitch at 7.47s into the file:

[20250423210926.zip](https://github.com/user-attachments/files/19884317/20250423210926.zip)

Has anyone looked for this on your own nodes? Since it's just some short quiet clicks it's often not noticeable during normal use but is more obvious in the telemetry speech since that audio is somewhat loud and very clear ie. no background noise.

Above recording was made with the archivedir function in wav format.

mkmer, 2025-04-24:
I haven't been able to create this on either of my Dell 3040 nodes, but I haven't spent a lot of time looking for it in the wav files either.  Ever since I built my radioless node, there has been a bit of "static" I attributed to my "cheap" hardware or speaker.  The clicking problem started with the kernel update back in (January?) but was addressed by "band-aid" kernel settings.
I plan on checking it out again once we get through a few pressing bugs.  

Your latest file looks more like a muted frame (it's long enough).  Definitely a head scratcher at the moment

jxmx, 2025-04-25:
I swapped out a Pi5 for a 3040 device. I then can do an `rpt playback` from the Asterisk CLI with `outstreamcmd = /bin/sh,-c,/usr/bin/lame -r -m m -s 8 --bitwidth 16 - /tmp/txcap.wav`. In general, I use a 1 kHz audio tone. Looking at the wave in Audacity, I don't see the nulls you're seeing nor can I hear them.

I'll do more testing to capture from USB next.

[3040_460180_1khz_outstream.zip](https://github.com/user-attachments/files/19912255/3040_460180_1khz_outstream.zip)

jxmx, 2025-04-25:
Further information about the setup is that I have no current IAX, SIP, or any other network connection. This is currently a standalone node 460180. Does this only occur when a network link is present?

jxmx, 2025-04-25:
> Allmon3 & AllScan

What this occurs do you have an open Allmon3 or Allscan window? If so, does the behavior cease if you have no open monitoring?

jxmx, 2025-04-25:
Connecting and monitoring Allmon3 does not cause the chop.

jxmx, 2025-04-25:
What are the specs of your 3040? Mine is:

* Four core Intel Atom x5-Z8350 @ 1.44 GHz
* 2 Gb RAM
* No WiFi
* 5v DC power

davidgsd, 2025-04-25:
This will happen even if nothing else is going on. ie. not connected to any other nodes, no web dashboard open. And yes all 3040's I know of have the same CPU and RAM specs. This occurs on both 5V and 12V 3040's, and on those with and without built-in WiFi.

The outstream command may be masking the issue because it seems to save to a very lossy gsm format, which will do something like an FFT and retain the 1KHz sine wave while dropping other harmonics and transients, so any short glitch will not show up as zeroes, and will be pretty well smoothed out. Thus the archivedir function in wav format should work much better.

jxmx, 2025-04-25:
I've changed over to using archivedir to capture the audio. A playback and RF receive are attached.

[20250425143249.zip](https://github.com/user-attachments/files/19914382/20250425143249.zip)

davidgsd, 2025-04-25:
Those look fine, thus the issue either just didn't happen in those couple of short tests or you have something in your configuration that somehow prevents the issue. In my nodes I'm not doing anything unusual in the configuration, and see this in both half and full duplex SimpleUSB configurations. (I don't often use USBRadio so haven't confirmed if that would make a difference.) You're running the latest main asl3? Are there any settings you have at non-default values in your .conf files? Thanks

jxmx, 2025-04-25:
This node is using a SimpleUSB adapter going to a Masters Communications RA-25 audio board. Nothing fancy. The configuration is "out of the box" as this is/was one of my upgrade test nodes. It's running standard Debian 12 amd64, kernel 6.1.0-33-amd64. asl3-asterisk is the current version - 22.2.0+asl3-3.4.4-1.deb12.

Later, I'll run some larger audio looping samples and see if runtime causes any accumulated defects.

jxmx, 2025-04-26:
I connected to the Dixie Trader Net for awhile (which is always a cornucopia of random traffic). I let it run for ~ 10 minutes. Here's the last audio file. I don't hear any glitches.

[20250425201410.zip](https://github.com/user-attachments/files/19918219/20250425201410.zip)

Did you try this on multiple 3040s? I wonder if there's some particular firmware thing with your system(s)?

davidgsd, 2025-04-26:
I've seen this on at least 10 3040's, with a wide variety of configurations. I do also install Allmon and AllScan which brings in all the python and php packages, asl3-update-nodelist, and avahi-daemon.

In modules.conf I enable
```
load = res_agi.so
load = res_speech.so
```

And for the playback of the IP address I have,
```
rpt.conf
startup_macro = *690
69 = autopatchup,context=my-ip,noct=1,farenddisconnect=1,dialtime=9000,quiet=1
```

extensions.conf
```
; *690: Say LAN IP Address
exten => 0,1,Wait(2)
exten => 0,n,AGI(getip.sh)
exten => 0,n,SayAlpha(${result})
exten => 0,n,Hangup
```

Where getip.sh =
```
#!/bin/bash
echo SET VARIABLE result `hostname -I | cut -f1 -d' '`
```
in /usr/share/asterisk/agi-bin/

Maybe something about one of these added modules/packages is enough to cause the issue. It seems to always happen within a few seconds on any Tx. Just in the IP address announce there's always a few glitches.

InterLinked1, 2025-04-26:
> I've seen this on at least 10 3040's, with a wide variety of configurations. I do also install Allmon and AllScan which brings in all the python and php packages, asl3-update-nodelist, and avahi-daemon.
> 
> In modules.conf I enable
> 
> ```
> load = res_agi.so
> load = res_speech.so
> ```

Unrelated, but I don't think you need `res_speech`, at least I don't see it being used in your example here below. (`SayAlpha` isn't part of `res_speech`.)

> And for the playback of the IP address I have, rpt.conf
> 
> ```
> startup_macro = *690
> 69 = autopatchup,context=my-ip,noct=1,farenddisconnect=1,dialtime=9000,quiet=1
> ```
> 
> extensions.conf
> 
> ```
> ; *690: Say LAN IP Address
> exten => 0,1,Wait(2)
> exten => 0,n,AGI(getip.sh)
> exten => 0,n,SayAlpha(${result})
> exten => 0,n,Hangup
> ```
>
> Where /usr/share/asterisk/agi-bin/getip.sh =
> 
> ```
> #!/bin/bash
> echo SET VARIABLE result `hostname -I | cut -f1 -d' '`
> ```

Honestly, you don't really need `res_agi` either, you could replace the AGI call with just:

`same => n,Set(result=${SHELL(hostname -I | cut -f1 -d)})`

Instead, you'd need `func_shell` for that.

jxmx, 2025-04-26:
> Maybe something about one of these added modules/packages is enough to cause the issue.

If you build a stock Debian install, just install the packages and use the default 1999 configuration do you have the same behaviors? My 3040 is a basic Debian 12 install with just the asl3 packages installed and then node changed and logged in.

davidgsd, 2025-04-26:
I installed Deb 12 on another 3040, and then installed asl3 and nothing else. I ran asl-menu, tried to set up node 1999 but it said that was already defined so I set up 1998 instead, and selected SimpleUSB. It then brings up the susbtune menu, but will not find the USB device. This happens often, even though the URI/UCI was plugged in the whole time (probably need to open another bug report on that). I then rebooted the 3040, and then the URI was properly recognized and the ID telemetry was played - and sure enough there are some glitches. Then a few seconds later the status light on the UCI stopped blinking. (Looks like a 2nd new bug report I could open.) devstr= is blank in simpleusb.conf. Has ASL stopped filling that in?

As you mentioned 1999 above and asl-menu said something about that already being defined I figured I should try that again and be sure to go with the absolute default configuration, since apparently doing anything at all results in the issue. I apt removed asl3, deleted the /etc/asterisk/ folder, rebooted, did an apt autoremove to be sure everything was cleared out, and then reinstalled asl3. That didn't do what it should have as none of the .conf files were written to /etc/asterisk/. Tried again with `sudo apt install --reinstall asl3` and still no conf files in etc/asterisk/. (Looks like a 3rd new bug report I could open.)

Will now reinstall debian on this so I can get to an absolute untouched starting point.

Allan-N, 2025-04-26:
>tried to set up node 1999 but it said that was already defined

Fix with https://github.com/AllStarLink/asl3-menu/pull/82

>devstr= is blank in simpleusb.conf. Has ASL stopped filling that in?

The `devstr` should be filled in when you "write" your settings from the tune menu

Allan-N, 2025-04-26:
I think `sudo apt install --reinstall asl3` processes just that one package and not the dependencies.  The config files are in the `asl3-asterisk-config` package.

davidgsd, 2025-04-26:
Allan, this is the usual thing I see when building nodes. I run asl-menu, enter the node# details, select duplex and channel driver, it opens the tune menu, but it doesn't actually find the device, and nothing can be done in the tune menu to find it. Only after I exit the menu and reboot the node, then is the USB interface finally recognized. And it's not consistent either. Sometimes it does find the USB interface when doing the set up in asl-menu, seems to be a 50/50 chance.  Sometimes just restarting asterisk is enough for it to find the USB interface, but often a reboot of the node is needed.

Allan-N, 2025-04-26:
The just-after-install issue may relate to the system needing a reboot for the `/etc/udev/rules.d/90-asl3.rules` to be applied.

davidgsd, 2025-04-26:
OK reinstalled deb12, and did the absolute minimal changes possible. Did absolutely nothing other than the following:
  cd /tmp
  wget https://repo.allstarlink.org/public/asl-apt-repos.deb12_all.deb
  sudo dpkg -i asl-apt-repos.deb12_all.deb
  sudo apt update
  sudo apt install asl3

Rebooted node. Did *not* run asl-menu

Added the following to rpt.conf:
archivedir = /tmp
archiveformat = wav

I have not changed a single setting in any other file.

I then did a *712 DTMF command to say the time. File is attached. There is a glitch at 0.720s into the file.

[20250426125617.zip](https://github.com/user-attachments/files/19924927/20250426125617.zip)

If anyone claims to be unable to reproduce this I would be happy to mail you this 3040.

davidgsd, 2025-04-26:
I also tried with USBRadio channel driver and that makes no difference. Attached is another example, where I did a DTMF *711 (ID). There are glitches in both the courtesy tone at 0.287s into the file, and in the Morse ID at 3.483s. 

[20250426135356.zip](https://github.com/user-attachments/files/19925079/20250426135356.zip)

jxmx, 2025-04-26:
> I then did a *712 DTMF command to say the time.

Specifically, HOW are you executing this command? Are you sending it over RF or calling `rpt func` from the Asterisk CLI. As you said above, you've been generating all the audio purely locally but this seems to involve a sound device and a radio

jxmx, 2025-04-26:
> That didn't do what it should have as none of the .conf files were written to /etc/asterisk/. Tried again with `sudo apt install --reinstall asl3` and still no conf files in etc/asterisk/. (Looks like a 3rd new bug report I could open.)

This is not a bug. If you uninstalled it but did not `apt purge` them, then Debian's dpkg is still tracking the files as "hand modified" which includes deleting them by hand. In order to reset the packages' configuration they way you managed the packages, you need to do:

```
apt install --reinstall -o Dpkg::Options::="--force-confask,confnew,confmiss" asl3-asterisk asl3-asterisk-config asl3-asterisk-modules
```

davidgsd, 2025-04-26:
It doesn't matter how you trigger the playback, there are always glitches. When the node powers on and plays the courtesy tone and ID there are glitches (which did not require any DTMF command). And if I do a DTMF command from a USB interface, there are glitches, and if I do a DTMF command from the CLI there are glitches. Here is the recording from when I turned it on just now. I did nothing other than turn it on. It ID'd after bootup and there is a glitch at 4.863s into the file.

[20250426164155.zip](https://github.com/user-attachments/files/19925819/20250426164155.zip)

jxmx, 2025-04-26:
This has me quite perplexed..... just grasping at straws, what USB device(s) do you have connected to the 3040s? Does the issue occur if you have no USB devices plugged in? I wonder if this is still a manifestation of the buffer under-runs in corner cases with certain types of hardware...

davidgsd, 2025-04-27:
Update from slack msgs:  Issue does **not** happen if no channel driver is enabled. The glitches are definitely a delay where the audio is shifted in time, as opposed to just zeroed samples. So the issue likely a result delays in USB I/O. 
[image]
Also confirmed the issue happens with both CM108B and CM108AH, and with both the USB 2.0 or 3.0 ports on the 3040.

davidgsd, 2025-04-27:
[image]
This confirms definitely that the audio is being shifted in time. Comparison of same "the time is..." audio but from 2 different files.

davidgsd, 2025-04-27:
Additional test notes:
* Restored a 3040 image from early Jan with **6.1.0-26**-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1.112-1 (2024-09-30) x86_64 GNU/Linux / Asterisk 20.11.0+asl3-3.2.0-2.deb12, and the audio has some the same sort of distortion, bit looks like the archive dir function did not support wav at that point. I notice the extension is WAV instead of wav, and it looks/sounds compressed, so transients/zeros are not as easy to confirm, but audio is clearly distorted.
*Turns out this happens on Beelink T5 Mini PCs also, but not to nearly the same extent. Parrot audio and asl-tts sound perfect on the T5 but there was one glitch in the ID. Tried about 5 tests and only heard one glitch. 
* Saw the same sort of occasional glitches on an ASL2 node (3040). Can't be sure it's the same thing because it appears the archivedir function does not save to uncompressed wav. But I recall now I have often heard faint little clicks in audio before, which isn't really noticeable in noisier signals but is more obvious in purer tones or tts. It seems to happen far less in ASL2 though and the audio overall is very clean.
* Tested on an RPi4 and no glitches there.
Conclusion: This seems to have been in ASL3 from the start, possibly in ASL2 to a lesser extent, but not on a RPi4, and not really an issue on a more powerful MiniPC (Celeron CPU vs. Atom). But the audio quality is definitely substantially degraded and distorted on 3040's. Maybe not noticeable on a noisier/lower quality RF signal, but pretty obvious on better quality signals if you listen carefully. Could be easily mistaken for a slightly overdriven mic. Hopefully you all have some ideas how to fix this.

Allan-N, 2025-04-27:
>but looks like the archivedir function did yet not support uncompressed wav then - extension is WAV instead of wav, and it looks/sounds compressed, so transients/zeros are not as easy to confirm, but audio is clearly distorted.

Correct.  We added the ability to set the format with [Issue #485](https://github.com/AllStarLink/app_rpt/issues/485), [PR #492](https://github.com/AllStarLink/app_rpt/pull/492).  Also, the .WAV extension is for gsm in a wav container.

jxmx, 2025-04-27:
I've updated the description of this issue.

davidgsd, 2025-04-27:
Looks like I have a fix for the issue. Not sure yet if it's a 100% fix, but it seems to be a solution so far. I turned of Intel SpeedStep and C-States options in the 3040 BIOS and all seems much better now. As the issue seemed CPU-related due to not occurring on a higher-end Mini PC I thought I would review the BIOS settings. And I know the SpeedStep stuff can also cause issues in Windows pro audio recording interfaces. Will do some further testing and confirm in the next couple days if I see any other issues. Recommended BIOS settings FYI:

[image]

Allan-N, 2025-05-07:
This issue was resolved with BIOS setting changes.

closed 2025-05-07 by Allan-N