AllStarLink/app_rpt #637: DTMF to remote node(s) not working

KenHorse, 2025-04-19:
The asterisk CLI (on the sending node/server) reports:

`[2025-04-15 17:16:14.203] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough 'A' on SimpleUSB/1102
[2025-04-15 17:16:14.331] DTMF[635]: channel.c:4019 __ast_read: DTMF begin 'B' received on SimpleUSB/1102
[2025-04-15 17:16:14.331] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough 'B' on SimpleUSB/1102
[2025-04-15 17:16:14.459] DTMF[635]: channel.c:3905 __ast_read: DTMF end 'B' received on SimpleUSB/1102, duration 128 ms
[2025-04-15 17:16:14.460] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin 'B' on SimpleUSB/1102
[2025-04-15 17:16:14.460] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough 'B' on SimpleUSB/1102
[2025-04-15 17:16:14.608] DTMF[635]: channel.c:4019 __ast_read: DTMF begin 'C' received on SimpleUSB/1102
[2025-04-15 17:16:14.608] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough 'C' on SimpleUSB/1102
[2025-04-15 17:16:14.758] DTMF[635]: channel.c:3905 __ast_read: DTMF end 'C' received on SimpleUSB/1102, duration 149 ms
[2025-04-15 17:16:14.758] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin 'C' on SimpleUSB/1102
[2025-04-15 17:16:14.758] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough 'C' on SimpleUSB/1102
[2025-04-15 17:16:14.858] DTMF[635]: channel.c:4019 __ast_read: DTMF begin '1' received on SimpleUSB/1102
[2025-04-15 17:16:14.858] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough '1' on SimpleUSB/1102
[2025-04-15 17:16:15.013] DTMF[635]: channel.c:3905 __ast_read: DTMF end '1' received on SimpleUSB/1102, duration 155 ms
[2025-04-15 17:16:15.013] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin '1' on SimpleUSB/1102
[2025-04-15 17:16:15.013] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough '1' on SimpleUSB/1102
[2025-04-15 17:16:15.141] DTMF[635]: channel.c:4019 __ast_read: DTMF begin '2' received on SimpleUSB/1102
[2025-04-15 17:16:15.141] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough '2' on SimpleUSB/1102
[2025-04-15 17:16:15.291] DTMF[635]: channel.c:3905 __ast_read: DTMF end '2' received on SimpleUSB/1102, duration 150 ms
[2025-04-15 17:16:15.291] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin '2' on SimpleUSB/1102
[2025-04-15 17:16:15.292] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough '2' on SimpleUSB/1102
[2025-04-15 17:16:15.398] DTMF[635]: channel.c:4019 __ast_read: DTMF begin '3' received on SimpleUSB/1102
[2025-04-15 17:16:15.398] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough '3' on SimpleUSB/1102
[2025-04-15 17:16:15.548] DTMF[635]: channel.c:3905 __ast_read: DTMF end '3' received on SimpleUSB/1102, duration 150 ms
[2025-04-15 17:16:15.548] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin '3' on SimpleUSB/1102
[2025-04-15 17:16:15.548] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough '3' on SimpleUSB/1102
[2025-04-15 17:17:32.193] DTMF[635]: channel.c:4019 __ast_read: DTMF begin 'A' received on SimpleUSB/1102
[2025-04-15 17:17:32.193] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough 'A' on SimpleUSB/1102
[2025-04-15 17:17:32.341] DTMF[635]: channel.c:3905 __ast_read: DTMF end 'A' received on SimpleUSB/1102, duration 148 ms
[2025-04-15 17:17:32.342] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin 'A' on SimpleUSB/1102
[2025-04-15 17:17:32.342] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough 'A' on SimpleUSB/1102
[2025-04-15 17:17:32.469] DTMF[635]: channel.c:4019 __ast_read: DTMF begin 'B' received on SimpleUSB/1102
[2025-04-15 17:17:32.469] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough 'B' on SimpleUSB/1102
[2025-04-15 17:17:32.597] DTMF[635]: channel.c:3905 __ast_read: DTMF end 'B' received on SimpleUSB/1102, duration 127 ms
[2025-04-15 17:17:32.597] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin 'B' on SimpleUSB/1102
[2025-04-15 17:17:32.597] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough 'B' on SimpleUSB/1102
[2025-04-15 17:17:32.726] DTMF[635]: channel.c:4019 __ast_read: DTMF begin 'C' received on SimpleUSB/1102
[2025-04-15 17:17:32.726] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough 'C' on SimpleUSB/1102
[2025-04-15 17:17:32.853] DTMF[635]: channel.c:3905 __ast_read: DTMF end 'C' received on SimpleUSB/1102, duration 127 ms
[2025-04-15 17:17:32.853] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin 'C' on SimpleUSB/1102
[2025-04-15 17:17:32.853] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough 'C' on SimpleUSB/1102
[2025-04-15 17:17:33.004] DTMF[635]: channel.c:4019 __ast_read: DTMF begin '1' received on SimpleUSB/1102
[2025-04-15 17:17:33.004] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough '1' on SimpleUSB/1102
[2025-04-15 17:17:33.152] DTMF[635]: channel.c:3905 __ast_read: DTMF end '1' received on SimpleUSB/1102, duration 148 ms
[2025-04-15 17:17:33.152] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin '1' on SimpleUSB/1102
[2025-04-15 17:17:33.152] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough '1' on SimpleUSB/1102
[2025-04-15 17:17:33.253] DTMF[635]: channel.c:4019 __ast_read: DTMF begin '2' received on SimpleUSB/1102
[2025-04-15 17:17:33.253] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough '2' on SimpleUSB/1102
[2025-04-15 17:17:33.408] DTMF[635]: channel.c:3905 __ast_read: DTMF end '2' received on SimpleUSB/1102, duration 155 ms
[2025-04-15 17:17:33.409] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin '2' on SimpleUSB/1102
[2025-04-15 17:17:33.409] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough '2' on SimpleUSB/1102
[2025-04-15 17:17:33.536] DTMF[635]: channel.c:4019 __ast_read: DTMF begin '3' received on SimpleUSB/1102
[2025-04-15 17:17:33.536] DTMF[635]: channel.c:4030 __ast_read: DTMF begin passthrough '3' on SimpleUSB/1102
[2025-04-15 17:17:33.686] DTMF[635]: channel.c:3905 __ast_read: DTMF end '3' received on SimpleUSB/1102, duration 149 ms
[2025-04-15 17:17:33.686] DTMF[635]: channel.c:3956 __ast_read: DTMF end accepted with begin '3' on SimpleUSB/1102
[2025-04-15 17:17:33.686] DTMF[635]: channel.c:3994 __ast_read: DTMF end passthrough '3' on SimpleUSB/1102
`

However, neither of the 2 nodes on the network (which are connected to Node 1102) shows any receipt of DTMF at all with their CLI.

This used to work but only recently noticed it was no longer.

Asterisk 22.2.0+asl3-3.4.2-1.deb12

InterLinked1, 2025-04-19:
Is the DTMF log level enabled on the receiving node?
Digits are no longer sent to the verbose level like they used to be.

mkmer, 2025-04-19:
It would be VERY helpful if you could figure out when it broke (what is the last version that "worked") - We have been working on a lot of problems, and having a place to start would help (a lot).

KenHorse, 2025-04-19:
DTMF logging is enabled. I can hear DTMF being sent by the remote node on other nodes on the network but all the local node's CLI shows is the remote node is keyed. 

/var/log/asterisk/messages.log shows no entries at all referencing DTMF.

As for when it happened exactly, my best guess is a few weeks ago. If there are specific tests you want me to run, I can. I am very Linux savvy and not a noob to ASL

KenHorse, 2025-04-19:
Update. I created a script in /etc/asterisk/custom and made it executable. In rpt.conf (under the [functions] stanza), I placed

`882 = cmd,/etc/asterisk/custom/myscript.sh
`
I can run the script from the bash prompt and it runs fine

Restarted asterisk but this command is ignored. Other DTMF commands, such as *712 (tell time) work fine.

Allan-N, 2025-04-19:
I'm going to assume that this new script is being used to check if your DTMF commands are being processed.

What does the script do?  Did (or can) you add a command early in the script that will tell you that the command executed (e.g. "date >> /tmp/yes-i-ran")

You mentioned testing the script from the bash prompt (as the user asterisk) and that this was OK.  Did you also try to run the command from the Asterisk CLI?

and if this doesn't help move the needle then can you attach your `rpt.conf` and `myscript.sh` files?

KenHorse, 2025-04-19:
The script ssh's into my HUB node to force a connect. To wit:

`#!/bin/bash

/usr/bin/ssh asterisk@10.147.156.98 "/usr/sbin/asterisk -rx 'rpt fun 1100 *81354326

'"`

If I run the following from the asterisk CLI:

`rpt fun 1100 *81354326`

It does not work

And yes, I can my script from the bash prompt and node 1100 connects properly to 1102

Attached it my rpt.conf

[rpt.zip](https://github.com/user-attachments/files/19822768/rpt.zip)

Allan-N, 2025-04-19:
To check "if" the script is executing I'd like to suggest you update your script to be :
```
#!/bin/bash
date >> /tmp/myscript-executed
/usr/bin/ssh asterisk@10.147.156.98 "/usr/sbin/asterisk -rx 'rpt fun 1100 *81354326'"
```
Then, regardless of whether the `ssh` worked you can at least know that asterisk started the script by checking the content of the temp file that should include a time stamp on each execution.

Also, since you are looking at a script that uses the `ssh` command I'll ask if you are having a repeat of the issues discussed in https://community.allstarlink.org/t/cmd-not-working-in-asl3/22681/13

KenHorse, 2025-04-19:
Yea, I remember my previous mistake and that's why manually running the script verifies the ssh connect is good. 

I was thinking about piping something in the script to a file so I verified it actually ran before I ran it manually and it never missed a beat. But for the sake of argument, I've added your pipe and then ran the script, manually and via a DTMF call. 

The results are:

Manually running myscript results in /tmp/myscript-executed - Sat Apr 19 01:59:02 PM PDT 2025

Via DTMF - no further entry in the file

mkmer, 2025-04-19:
You may already have done this... the script when triggered from asterisk is run as the user "asterisk".  Make sure the script either belongs to user asterisk, or it's executable by all.

KenHorse, 2025-04-19:
As I stated above, I've run it as asterisk (and root for that matter)

KenHorse, 2025-04-19:
Maybe this will help track things down.

I reverted to Asterisk 22.2.0+asl3-3.3.0-6.deb12 and my script(s) run fine when called via DTMF. (the joy of Proxmox snapshot backups)

Allan-N, 2025-04-19:
Ding!

You wrote that your script included :
```
/usr/bin/ssh asterisk@10.147.156.98 "/usr/sbin/asterisk -rx 'rpt fun 1100 *81354326'"
```

That command is saying "ssh" to the remote system LOGGING IN AS THE USER "asterisk" and execute the `/usr/sbin/asterisk` command.  Without relaxing the asterisk permissions on the remote system don't you really want the command to exec `/usr/sbin/asterisk` as "root"?

Do you get the expected behavior if you change that line to :
```
/usr/bin/ssh root@10.147.156.98 "/usr/sbin/asterisk -rx 'rpt fun 1100 *81354326'"
```

KenHorse, 2025-04-19:
Way ahead of you

-rwxr-xr-x 1 root root 3439208 Mar  6 15:08 /usr/sbin/asterisk

Allan-N, 2025-04-20:
The permissions on the file do not control how the command is executed.  They only determine who can access the file (and how the file can be accessed).

KenHorse, 2025-04-20:
I get that. Are you saying that user asterisk, running asterisk, isn't correct? I thought that was a big change in ASL3, where asterisk runs asterisk not root

Allan-N, 2025-04-20:
Sorry, my bad.  I knew there was a issue with non-"root" logins not being able to exec `asterisk -x ...` because the file permissions on the control socket (`/var/run/asterisk/asterisk.ctl`) limits who can write. What I forgot was that the "asterisk" user should be OK as it's the owner of the file.

and I thought I had this one figured out ...

Allan-N, 2025-04-20:
Just checking ...
- the host at IP `10.147.156.98` hosts node `1100` ?
- the host at IP `10.147.156.98` hosts supports exec'ing the `*81354326` function (`813 = ...` is in this hosts rpt.conf/[functions]) ?

KenHorse, 2025-04-20:
- Correct
- Correct

mkmer, 2025-04-20:
I can use your exact command line to connect and disconnect a node (changing the node and host of course) using my "personal" user that is added to the asterisk group AND asterisk is setup to allow "others".  I can not run the same command with the asterisk user. 
It must be some permission issue ... not sure which one at the moment.
`ssh asterisk@wyse.local "/usr/sbin/asterisk -rx 'rpt fun 572920 *81329972'"` - does not work
`ssh mkasper@wyse.local "/usr/sbin/asterisk -rx 'rpt fun 572920 *81329972'"` - does work

InterLinked1, 2025-04-20:
Just want to clarify since the scope here has changed significantly, it seems like DTMF actually works fine and this is an unrelated issue of a custom script not executing (which would be triggered by DTMF), or something like that? (And if it is suspected to be DTMF, debugging the script is not the way to debug that)

You can use `iax2 set debug on` or the `FRAME_TRACE` function to confirm if DTMF is really coming across the wire. If there is, there's likely no DTMF issue here.

KenHorse, 2025-04-20:
Synopsis:

- Running the script from DTMF doesn't work while running other DTMF commands do (for example, *712 speaks the time).

- Running the script manually as user asterisk works

InterLinked1, 2025-04-20:
> Synopsis:
> 
>     * Running the script from DTMF doesn't work while running other DTMF commands do (for example, *712 speaks the time).
> 
>     * Running the script manually as user asterisk works
> 
>     * I reverted back to Asterisk 22.2.0+asl3-3.3.0-6.deb12 and DTMF runs my script.

Okay, so it's not a DTMF issue then, but an issue with that particular command. Thanks for clarifying.

Allan-N, 2025-04-20:
Or is there still a DTMF issue and you are trying to use the asterisk command(s) as a workaround?

@KenHorse, can you share/repeat a bit of the big picture like "I have an RF radio (HT) that I want to use to control a node/repeater/whatever.  I want/need to be able to issue DTMF tones that will be received by the repeater/node XXXX but need to be passed to a different repeater/node YYYY.   And the DTMF tones need to be processed by Asterisk/app_rpt on YYYY directly?  or do you need the audio tones to go into YYYY and then audio out to a non-ASL repeater controller?  or what?"

KenHorse, 2025-04-20:
Breaking it down to its simplest form, I run a multi-private node system which also has a public node linked in. 

Private node 1100 is the hub for 2 remote repeaters (nodes 1101 & 1102). Each repeater site also uses an external repeater controller and ASL is connected to it via a RIM Lite. 

Users of the 1102 repeater enter a DTMF sequence that runs a Macro within the controller, which then sends a DTMF sequence to ASL, waits 2 seconds and send a 2nd DTMF sequence. Node 1102 then runs scripts that ssh directly to node 1100 which in turn carries out the commands in the script.

I hope this explains it well enough.

Allan-N, 2025-04-20:
>I hope this explains it well enough.

Yes, the explanation helped.

Also, yesterday you attached an rpt.conf file from what looks to be node 1102.  In looking over that file I noticed that all of the [macro]'s are commented out.  Just wanted to be sure you were aware of this.

KenHorse, 2025-04-21:
Yes, I don't use Macros on that node - the controller handles all that.

mkmer, 2025-04-21:
Well - maybe not.  unsigned int:1 -> still resolves to 1.

KenHorse, 2025-04-21:
I copied the ssh keys from node 1102 to node 1100 so no password is either needed nor asked for when the script is run

mkmer, 2025-04-21:
Into the asterisk user "home" -> `/var/lib/asterisk/.ssh`  
Maybe they are not in the updated version so the restore proxmox works?

I'm unable to reproduce at this point.  I believe I have implemented a similar config as you are running and it appears to work for me.
Are you seeing similar behavior - aka DTMF does execute the script creating a time/date stamp but no ssh? Or no script execution at all?