AllStarLink/app_rpt #639: Rename CLI commands with non-conforming command names

InterLinked1, 2025-04-19:
Many of the CLI commands in all of the module names have cryptic/terse command names that don't conform to the Asterisk CLI command naming conventions: https://docs.asterisk.org/Operation/Asterisk-Command-Line-Interface/CLI-Syntax-and-Help-Commands/

Many of them should be renamed, including every CLI command that currently only consists of two words. For compatibility, aliases for the old command names can be defined in `cli_aliases.conf`.

As an example, `rpt localnodes` should be renamed to `rpt show <something>`, maybe just `rpt show nodes`, possibly with other modifiers accepted.

Allan-N, 2025-04-19:
Dup of (or related to) https://github.com/AllStarLink/app_rpt/issues/131 ?

InterLinked1, 2025-04-19:
Yup - I still had this in my personal notes and was trying to clean some stuff up and didn't find it in a search. Closing as duplicate.

closed 2025-04-19 by InterLinked1