AllStarLink/app_rpt #645: ASL3 + echolink crashing

okcrum, 2025-04-23:
User is reporting crashes of Asterisk 22.2.0+asl3-3.4.3-1.deb12 on a busy system. Backtrace and message logs follow.

[messages.txt](https://github.com/user-attachments/files/19863820/messages.txt)
[core-asterisk-2025-04-22T20-13-51Z-thread1.txt](https://github.com/user-attachments/files/19863819/core-asterisk-2025-04-22T20-13-51Z-thread1.txt)
[core-asterisk-2025-04-22T20-13-51Z-locks.txt](https://github.com/user-attachments/files/19863822/core-asterisk-2025-04-22T20-13-51Z-locks.txt)
[core-asterisk-2025-04-22T20-13-51Z-info.txt](https://github.com/user-attachments/files/19863821/core-asterisk-2025-04-22T20-13-51Z-info.txt)
[core-asterisk-2025-04-22T20-13-51Z-full.txt](https://github.com/user-attachments/files/19863818/core-asterisk-2025-04-22T20-13-51Z-full.txt)
[core-asterisk-2025-04-22T20-13-51Z-brief.txt](https://github.com/user-attachments/files/19863817/core-asterisk-2025-04-22T20-13-51Z-brief.txt)

Reference: ASL forum thread [ASL3-Ecolink asterisk Crashing](https://community.allstarlink.org/t/asl3-ecolink-asterisk-crashing/23164)[](https://community.allstarlink.org/t/asl3-ecolink-asterisk-crashing/23164)

okcrum, 2025-04-23:
I have another backtrace from my production hub with a memory related crash in chan_echolink on do_new_call(). I have been unable to reproduce it.

[messages.log](https://github.com/user-attachments/files/19871054/messages.log)
[core-asterisk-2025-04-21T20-49-14Z-thread1.txt](https://github.com/user-attachments/files/19871057/core-asterisk-2025-04-21T20-49-14Z-thread1.txt)
[core-asterisk-2025-04-21T20-49-14Z-locks.txt](https://github.com/user-attachments/files/19871053/core-asterisk-2025-04-21T20-49-14Z-locks.txt)
[core-asterisk-2025-04-21T20-49-14Z-info.txt](https://github.com/user-attachments/files/19871052/core-asterisk-2025-04-21T20-49-14Z-info.txt)
[core-asterisk-2025-04-21T20-49-14Z-full.txt](https://github.com/user-attachments/files/19871055/core-asterisk-2025-04-21T20-49-14Z-full.txt)
[core-asterisk-2025-04-21T20-49-14Z-brief.txt](https://github.com/user-attachments/files/19871056/core-asterisk-2025-04-21T20-49-14Z-brief.txt)

InterLinked1, 2025-04-24:
> Just wanted to acknowledge we are looking at this. Any more info or crashes may be helpful. It's looking like one of the more difficult ones to figure out. PR [#646](https://github.com/AllStarLink/app_rpt/pull/646) is potentially related.

I don't think #646 is related. Memory leaks don't lead to memory corruption.
Based on the second crash and looking at the first, I don't think `ast_str_create` is necessarily implicated either, usually corruption happens sometimes after the offense was committed, and the backtrace thus won't clue you into the actual culprit.

It may be worth running the system under valgrind if possible, but that's very resource intensive, so I understand if that may not be feasible.

mkmer, 2025-04-25:
> I don't think [#646](https://github.com/AllStarLink/app_rpt/pull/646) is related. Memory leaks don't lead to memory corruption. Based on the second crash and looking at the first, I don't think [...]

I don't think it is either, that's why I unlinked it. That way it doesn't close the issue.  I wanted to tag it so we can "remember" what has been done so far.

Can you confirm that the channel private data says put when we masquerade a channel?  I did some digging and it looks like it does, but I'm not 100% sure.  IF we get a new pointer to the private data, the one stored in the channel driver could be causing the corruption.

InterLinked1, 2025-04-25:
> I don't think it is either, that's why I unlinked it. That way it doesn't close the issue. I wanted to tag it so we can "remember" what has been done so far.
>
> Can you confirm that the channel [...]

When a channel is masqueraded, pretty much all the guts of a channel are swapped out into a new channel structure and the name of the original channel is changed to have a Surrogate prefix, since its only purpose is to get killed off imminently. You can see the full masquerade operation here: https://github.com/asterisk/asterisk/blob/576f6bec3d9e136fa5672b6c80143540f2166bbf/main/channel.c#L6879

mkmer, 2025-04-25:
Ah - https://github.com/asterisk/asterisk/blob/576f6bec3d9e136fa5672b6c80143540f2166bbf/main/channel.c#L7010C10-L7010C30
Just copy the pointer and leave it alone - that rules out a bad private data pointer in here.

okcrum, 2025-04-25:
Running valgrind on an RPi 5 hub configured with EL. Here is output for ASL 3.4.3-1 (no crash, no valgrind summary):
[valgrind.out.3.4.3-1.txt](https://github.com/user-attachments/files/19916458/valgrind.out.3.4.3-1.txt)
I am collecting data from 3.4.4-1 ATM, will post it later.

InterLinked1, 2025-04-25:
> Running valgrind on an RPi 5 hub configured with EL. Here is output for ASL 3.4.3-1 (raw, no crash, no valgrind summary): [valgrind.out.3.4.3-1.txt](https://github.com/user-attachments/files/19916458/valgrind.out.3.4.3-1.txt) I am collecting data from 3.4.4-1 ATM, will post it later.

This looks pretty useful. Looks like it could possibly be what's going on here - and in any case, there are definite issues that should be resolved. Thanks for doing that!

okcrum, 2025-04-25:
NP. You were right about no leaks, but I was running Mike's recent ast_free() patch. I'll clean up the escape sequences so the next one will be more readable and hopefully complete with a crash. The heap creep might just impact operation eventually.

okcrum, 2025-04-25:
Here's valgrind output from 3.4.4-1. Things went south when I started changing connections (likely a result of corrupt heap data) and asterisk aborted.
[valgrind.out.3.4.4-1.txt](https://github.com/user-attachments/files/19917961/valgrind.out.3.4.4-1.txt)

Allan-N, 2025-04-26:
>Things went south when I started changing connections

Can you expand on this?

okcrum, 2025-04-26:
> > Things went south when I started changing connections
> 
> Can you expand on this?

I  had the hub  connected in permanent Transceive mode to one other node, and had one incoming EL connection to the hub. When I changed the permanent outbound connection to permanent Monitor, kaboom.

Here's a simpler scenario. All these are the same system, an RPi 5 hub with EL and simpleusb (so I can plug in a USB headset to monitor) . I set up two permanent monitor connections, then connected to the hub with EL client, kaboom. That trace follows.
[valgrind.out.3.4.4-1a.txt](https://github.com/user-attachments/files/19918224/valgrind.out.3.4.4-1a.txt)

Allan-N, 2025-04-26:
> I set up two permanent Monitor connections

What nodes were you monitoring?

okcrum, 2025-04-26:
559830 and 46005

okcrum, 2025-04-26:
Scenario for this crash:
1) make permanent Monitor connections to two nodes (in my test the nodes above)
2) connect with EL proxy client
This appears to be reproducible.

[messages.log](https://github.com/user-attachments/files/19918539/messages.log)
[core-asterisk-2025-04-26T00-48-38Z-thread1.txt](https://github.com/user-attachments/files/19918537/core-asterisk-2025-04-26T00-48-38Z-thread1.txt)
[core-asterisk-2025-04-26T00-48-38Z-locks.txt](https://github.com/user-attachments/files/19918540/core-asterisk-2025-04-26T00-48-38Z-locks.txt)
[core-asterisk-2025-04-26T00-48-38Z-info.txt](https://github.com/user-attachments/files/19918538/core-asterisk-2025-04-26T00-48-38Z-info.txt)
[core-asterisk-2025-04-26T00-48-38Z-full.txt](https://github.com/user-attachments/files/19918535/core-asterisk-2025-04-26T00-48-38Z-full.txt)
[core-asterisk-2025-04-26T00-48-38Z-brief.txt](https://github.com/user-attachments/files/19918536/core-asterisk-2025-04-26T00-48-38Z-brief.txt)

Allan-N, 2025-04-26:
FIrst off.  THANK YOU for the steps to repro.  This greatly simplifies debugging (and, FYI, you only need to be monitoring "1" node with many links).

I see a change that was made between 3.2.0 and 3.3.0 that may be a (the?) source of trouble.  But, it's late for me and best to debug tomorrow when I'm awake.

To those who like looking at code, in `el_text()` it looks like we dropped a `k = strlen(pkt);` statement yet we are still using the value `k`.

okcrum, 2025-04-26:
YW. I've never been a _tester_ per se but I have learned with nasty bugs to hang on to the scenario so _I_ can test it. Once I understand asterisk I can help a bit more perhaps.

mkmer, 2025-04-28:
What "version" of code are you running for the last few tests? Is it the 3.4 branch or master branch with Allan's update? 
Fwiw, there are some bugs in the master right now, and maybe that's part of the second set of crashes... (Maybe)

okcrum, 2025-04-29:
3.4.4-1 master, with Allan's chan_echolink.c patch.