AllStarLink/app_rpt #648: app_rpt: Telemery appears to spuriously key network TX

jxmx, 2025-04-23:
This is a follow-up to the #devel thread in Slack on spurious key-ups of nodes. What I am perceiving to be happening is as follows:

1. Non-RF network traffic comes into the node from a link
2. The traffic completes and the network traffic "hangs up"
3. Within 1 second, the previously-idle node begins the process to ID
4. The node is keying up the network links momentarily as part of the telemetry process

```
[2025-04-23 17:02:06.377] DEBUG[209687]: app_rpt/rpt_telemetry.c:1060 rpt_tele_thread: Requested channel DAHDI/pseudo-764362387
[2025-04-23 17:02:06.377] DEBUG[209687]: app_rpt/rpt_call.c:32 rpt_disable_cdr: No CDR present on DAHDI/pseudo-764362387
[2025-04-23 17:02:06.377] DEBUG[209687]: app_rpt/rpt_bridging.c:374 dahdi_conf_add: Channel DAHDI/pseudo-764362387 joining conference 1021
    -- Event on node 460181 doing rpt command cop,62,GPIO4:1 for condition c|t|RPT_RXKEYED
    -- Event on node 460181 doing rpt command cop,62,GPIO4:0 for condition c|f|RPT_RXKEYED
[2025-04-23 17:02:07.014] DEBUG[209689]: app_rpt/rpt_telemetry.c:1060 rpt_tele_thread: Requested channel DAHDI/pseudo-2065594029
[2025-04-23 17:02:07.014] DEBUG[209689]: app_rpt/rpt_call.c:32 rpt_disable_cdr: No CDR present on DAHDI/pseudo-2065594029
[2025-04-23 17:02:07.014] DEBUG[209689]: app_rpt/rpt_bridging.c:374 dahdi_conf_add: Channel DAHDI/pseudo-2065594029 joining conference 1021
    -- Hungup 'DAHDI/pseudo-764362387'
    -- Hungup 'DAHDI/pseudo-2065594029'
[2025-04-23 17:02:11.621] DEBUG[209694]: app_rpt/rpt_telemetry.c:1060 rpt_tele_thread: Requested channel DAHDI/pseudo-1574811709
[2025-04-23 17:02:11.621] DEBUG[209694]: app_rpt/rpt_call.c:32 rpt_disable_cdr: No CDR present on DAHDI/pseudo-1574811709
[2025-04-23 17:02:11.621] DEBUG[209694]: app_rpt/rpt_bridging.c:374 dahdi_conf_add: Channel DAHDI/pseudo-1574811709 joining conference 1021
[2025-04-23 17:02:11.621] DEBUG[209694]: app_rpt/rpt_channel.c:57 wait_interval: Delay interval = 500
[2025-04-23 17:02:12.121] DEBUG[209694]: app_rpt/rpt_channel.c:61 wait_interval: Delay complete
    -- Hungup 'DAHDI/pseudo-1574811709'
```

When I get some time, I will attempt to simulate this with network playbacks in a controlled environment.

InterLinked1, 2025-04-23:
> When I get some time, I will attempt to simulate this with network playbacks in a controlled environment.

This sounds almost like a test case to me. If we're able to formalize that as part of our tests, that would be great! Those would prevent a regression in the future.

jxmx, 2025-04-23:
So it's not simply a matter of it keying up for every telemetry. That isn't repeatable.

mkmer, 2025-04-24:
What version did this start in?
Did the telemetry say "anything" - > based on the log above, there were 3 calls that created 3 telemetry threads.  Each of those threads should have had "something" to say or play on their respective channels.
In general, if we have a telemetry thread, it keys up the repeater.

jxmx, 2025-04-24:
Basic on anecdotal and unscientific observation, the key-up of the network appears to be momentary with no audio sent.

mpannen1979, 2025-05-01:
jxmx:

Does this particular allstar node have any of the USB signaling set to active high?

jxmx, 2025-05-01:
No, it's all usbinvert. I've tried purposefully recreating this but I cannot. It's very annoyingly spurious.

Allan-N, 2025-05-07:
With https://github.com/AllStarLink/app_rpt/pull/678, I am adding some telemetry logging.