AllStarLink/app_rpt #659: Consolidate functions duplicated between app_rpt modules

Allan-N, 2025-04-26:
There are a number of functions in app_rpt, app_gps, and the channel drivers where we use multiple copies of the same source code.  It would be nice to have a single version of these functions that can be used by all, ideally without needing to compile multiple copies.

For example:
- `explode_string()`
- `finddelim()`
- `time_monotonic()`, `rpt_time_monotonic()`, ...

What is the asterisk equivalent of creating a library?

Bonus points if we can set things up so that the same library (or source files) can be used by simpleusb-tune-menu.c (e.g. `explode_string()`)

InterLinked1, 2025-04-26:
> There are a number of functions in app_rpt, app_gps, and the channel drivers where we use multiple copies of the same source code. It would be nice to have a single version of these functions that [...]

Just a module that exports global symbols, like `res_usb_radio` does currently.
However, such modules typically provide broad functionality used by a variety of modules. Utility functions generally go in the core.

> Bonus points if we can set things up so that the same library (or source files) can be used by simpleusb-tune-menu.c (e.g. `explode_string()`)

This is unlikely to be possible.

While we are looking at this, it's probably worth reviewing each of these functions and seeing if they can be simplified in any way or if we can leverage existing Asterisk utility functions. There are probably some that didn't exist when `app_rpt` was written.

Allan-N, 2025-04-26:
>Just a module that exports global symbols, like res_usb_radio does currently.

Can we export global symbols from `app_rpt.so` that would be available to our channel drivers?

InterLinked1, 2025-04-26:
> > Just a module that exports global symbols, like res_usb_radio does currently.
> 
> Can we export global symbols from `app_rpt.so` that would be available to our channel drivers?

Technically, yes, but no, this should not be done.