AllStarLink/app_rpt #666: app_rpt: Need to identify frames with f->data.ptr==NULL but f->samples>0

Allan-N, 2025-04-29:
As a followup to https://github.com/AllStarLink/app_rpt/pull/665, we need to understand why we are getting frames to process with no data (`f->data.ptr == NULL`) and a non-zero sample count (`f->samples > 0`).  Once identified, we need to resolve the conflict.

mkmer, 2025-04-29:
The suspect frame:
Samples > datalen
offset = 64 (probably shouldn't be)

```
$xx = { frametype = AST_FRAME_VOICE,
        subclass = {integer = 0, {format = 0x5556853f28, topology = 0x5556853f28}, frame_ending = 0},
        datalen = 0,
        samples = 160,
        mallocd = 1,
        mallocd_hdr_len = 222,
        offset = 64,
        src = 0x0,
        data = {ptr = 0x0, uint32 = 0, pad = "\000\000\000\000\000\000\000"},
        delivery = {tv_sec = 1745930822, tv_usec = 19338},
        frame_list = {next = 0x0},
        flags = 0,
        ts = 0,
        len = 0,
        seqno = 0,
        stream_num = 0
      }
```

Allan-N, 2025-04-29:
>offset = 64 (probably shouldn't be)

#define AST_FRIENDLY_OFFSET     64

mkmer, 2025-04-29:
> > offset = 64 (probably shouldn't be)
> 
> #define AST_FRIENDLY_OFFSET 64

Except it is populated without a data pointer.

Allan-N, 2025-04-30:
FYI :
```
#define AST_MALLOCD_HDR     (1 << 0)
```

mkmer, 2025-04-30:
This may be the source of our "bad" frame:
https://github.com/asterisk/asterisk/blob/754dea319e1d9113a6db01a921e8e424c5c71ffa/main/translate.c#L556-L562

okcrum, 2025-04-30:
An observation about the workaround:
If monitoring node A and node B; volume drops when both are keyed. When one node unkeys, volume returns to normal. So now we know what is supposed to happen when we get a bad frame.

okcrum, 2025-05-01:
The only other scrap of data I have on this is this valgrind excerpt from the crash:

==160180== Invalid read of size 2
==160180==    at 0x232EF0: ast_slinear_saturated_multiply_float (utils.h:493)
==160180==    by 0x232EF0: ast_frame_adjust_volume_float (frame.c:828)
==160180==    by 0xEA793DF: process_link_channels (app_rpt.c:4255)
==160180==    by 0xEA7CD07: rpt (app_rpt.c:5304)
==160180==    by 0x301A7F: dummy_start (utils.c:1607)
==160180==    by 0x59EEE9F: start_thread (pthread_create.c:442)
==160180==    by 0x5A57B1B: thread_start (clone.S:79)
==160180==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==160180== 
==160180== 
==160180== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==160180==  Access not within mapped region at address 0x0
==160180==    at 0x232EF0: ast_slinear_saturated_multiply_float (utils.h:493)
==160180==    by 0x232EF0: ast_frame_adjust_volume_float (frame.c:828)
==160180==    by 0xEA793DF: process_link_channels (app_rpt.c:4255)
==160180==    by 0xEA7CD07: rpt (app_rpt.c:5304)
==160180==    by 0x301A7F: dummy_start (utils.c:1607)
==160180==    by 0x59EEE9F: start_thread (pthread_create.c:442)
==160180==    by 0x5A57B1B: thread_start (clone.S:79)
==160180==  If you believe this happened as a result of a stack
==160180==  overflow in your program's main thread (unlikely but
==160180==  possible), you can try to increase the size of the
==160180==  main thread stack using the --main-stacksize= flag.
==160180==  The main thread stack size used in this run was 8388608.
==160180== 
==160180== HEAP SUMMARY:
==160180==     in use at exit: 21,461,215 bytes in 228,703 blocks
==160180==   total heap usage: 7,504,756 allocs, 7,276,053 frees, 2,015,244,082 bytes allocated

mkmer, 2025-05-01:
https://github.com/asterisk/asterisk/issues/1230

mkmer, 2025-06-10:
> [asterisk/asterisk#1230](https://github.com/asterisk/asterisk/issues/1230)
> 
> We shall see what happens with that...

Has been merged.  Should be "fixed".

mkmer, 2025-07-27:
I'm 99.9% sure that was the location of the "problem" we were experiencing.  At a minimum, we no properly handle these frames if they show up.

closed 2025-07-27 by mkmer