AllStarLink/app_rpt #670: Callsign data not being displayed in Echolink user app while connected to Echolink chan_echolink based nodes

k2cb, 2025-04-30:
Recently transitioned from running Echolink on a dedicated PC, to now using chan_echolink running on an ASL node, which is then connected to a repeater controller. 

When connecting to the Echolink Windows based node, users can see the callsigns of all connected stations.  Now, when connecting to the chan_echlolink based node, users now only see the Echolink node numbers, just like ASL nodes, and not the actual callsigns of the other connected stations.

This is an issue for large repeater systems where during nets etc,  we can have as many as 20 or more stations connected to the Echolink node.  Net control operators usually log into the Echolink node so they can see who is connected real-time.

Since switching over, we have had numerous inquires as to why the users no longer see the callsigns.

Can this functionality be added to chan_echolink?

Eric
K2CB

Allan-N, 2025-04-30:
This sounds like you are asking for each EL node# to be looked up in the EL database before being added to a list that is shared from the server.

Q? do you have access to a system that is still running EL on a dedicated PC?  If so, could you attach a screen capture showing how an EL node is displayed when connecting to that [PC] system vs. a screen capture showing how the same EL node is displayed to an [ASL] system?

k2cb, 2025-05-01:
[image]
[image]

k2cb, 2025-05-01:
Hi Allan,

The first image W2NJR-R is while connected to the AllStar node running chan_echolink, and the second image W2NJR-L is while connected to the Echolink PC.

Thank you.

Eric
K2CB

Allan-N, 2025-05-01:
>The first image W2NJR-R is while connected to the AllStar node running chan_echolink, and the second image W2NJR-L is while connected to the Echolink PC.

Thank you for the images.  I had an idea (hope) that you were referring to the textual information reporting the linked nodes.  As it turns out, I was just reworked some of that code to fix a few problems.  I will look to see if I can easily (and efficiently) map the EL node # to the EL callsign.

k2cb, 2025-05-01:
That would be awesome.  
Thank you.

Eric
K2CB

Allan-N, 2025-05-01:
Note: just to set expectations, this is not at/near the top of my TODO list ... but it's also not at the bottom

Allan-N, 2025-06-05:
@k2cb

On Community, we just announced [2025.06.05 - Package Updates - Beta Channel](https://community.allstarlink.org/t/2025-06-05-package-updates-beta-channel/23347).  The new packages include the changes for https://github.com/AllStarLink/app_rpt/pull/675 that add the EL callsign to the status instead of the EL node.  If you are willing / able to install the beta then it would be good to know if those changes address this request.

k2cb, 2025-06-06:
I installed the beta package updates on two nodes this evening. So far so good. Seeing Echolink station callsigns now.

THANK YOU!

Eric
K2CB

Allan-N, 2025-06-06:
Great!

closed 2025-06-06 by Allan-N