AllStarLink/app_rpt #672: Remote base mode changing

wb2dyb, 2025-05-03:
In most recent (past ten years or so) there has been a command missing in rpt.conf [remote-functions] which is mode selection. This means FM, AM, LSB, USB. Having used the remote feature since around 2011 I just modified my rpt.conf in later years. However this should probably be included in the distributed rpt.conf. The added line should be:

*61 = remote,61 ;FM Mode
*62 = remote,62  ;USB Mode
*63 = remote,63 ;LSB Mode
*64 = remote,64 ;AM Mode

from the app_rpt.c file:

		case 6: /* MODE (FM,USB,LSB,AM) */
			if(strlen(digitbuf) < 1)
				break;

			if(!multimode)
				return DC_ERROR; /* Multimode radios only */

			switch(*digitbuf){
				case '1':
					split_freq(mhz, decimals, myrpt->freq); 
					m=atoi(mhz);
					if(m < 29) /* No FM allowed below 29MHz! */
						return DC_ERROR;
					myrpt->remmode = REM_MODE_FM;
					
					rpt_telemetry(myrpt,REMMODE,NULL);
					break;

				case '2':
					myrpt->remmode = REM_MODE_USB;
					rpt_telemetry(myrpt,REMMODE,NULL);
					break;	

				case '3':
					myrpt->remmode = REM_MODE_LSB;
					rpt_telemetry(myrpt,REMMODE,NULL);
					break;
				
				case '4':
					myrpt->remmode = REM_MODE_AM;
					rpt_telemetry(myrpt,REMMODE,NULL);
					break;
		
				default:
					return DC_ERROR;

George Csahanin W2DB

;

wb2dyb, 2025-05-03:
This is in the source, but it needs to be added to the rpt.conf file
GC

Allan-N, 2025-05-25:
Notes:
- from the source code, we need to review (identify) all of the available remote base commands
- need to update update the remote base content in `rpt.conf`
- need to update [ASL3 Manual : Remote Base Nodes](https://allstarlink.github.io/adv-topics/remotebase)

closed 2025-05-28 by Allan-N