AllStarLink/app_rpt #676: 'Disconnect Storms' and 'Unwanted TX Keyup' with ASL3 3.4.1+ hub node and HamVOIP clients

mpannen1979, 2025-05-06:
I tried version 3.4.3 of ASL3 for a few weeks. I noticed that quite frequently many of the connected clients to our hub node would simultaneously drop out of the connected list on allmon3 and then reconnect (for those with 'permanent' connections anyway).  I will call this behavior a ‘hang up storm’.  This would happen at seemingly random times, sometimes many successively minutes apart, sometimes 3 hours apart.  I was not able to correlate this with anything in particular. I did not see any suspicious messages in the asterisk cli, or in the asterisk log file.  Our hub node typically hosts 20-29 clients, and runs on a pi3.  I had to revert to version 3.3.0 since this was too disruptive.

Since then I have made a test setup with 26 nodes connected:
node 452381 is the hub running version 3.4.5 of ASL3, on a pi3.
node 452384 is connected to the hub and has a simpleusb hotspot running version 3.4.5 of ASL3, on a pi3.
node 2495 is also connected to provide some traffic, and that node is running ASL3 3.3.0, on a pi3.
24 other nodes connected to the hub running hamvoip are set to private nodes in the range 1975-1998, on pi3's.

After a short period I noticed the ‘hang up storm’ from the 24 connected hamvoip nodes.
I also looked at allmon3 on the connected node 2495 mentioned above, and that node's allmon indicated my hub node 452381 was 'LAST RECV" at time 12:36:30, despite my hub node and all of the test nodes connected to it, were never in TX purposefully.  I will call this ‘unwanted TX key up’.

On a following test run a bit later I observed another ‘hang up storm’ and ‘unwanted TX key up’.  On one of the nodes that was part of the 'hang up storm', node 1998 with IP address 192.168.1.10, this was what that node's asterisk console showed at the time of hang up:

```
[May 3 14:58:16] ERROR[464] app_rpt.c: No link messages from [452381] in 46 seconds -- One way audio??? Forcing reconnect.
[May 3 14:58:16] WARNING[464] app_rpt.c: Node 1998: Reconnect Attempt to 452381 in process
```

I was able to tcpdump the IAX traffic from hub node 452381 with IP address 192.168.1.3, as well as tcpdump IAX traffic from node 1998, as well as debug asterisk console output from both nodes:

[1998-4569-192-168-1-10-tcpdump.zip](https://github.com/user-attachments/files/20052615/1998-4569-192-168-1-10-tcpdump.zip)
[1998-console.zip](https://github.com/user-attachments/files/20052614/1998-console.zip)
[452381-4569-192-168-1-3-tcpdump.zip](https://github.com/user-attachments/files/20052617/452381-4569-192-168-1-3-tcpdump.zip)
[452381-console.zip](https://github.com/user-attachments/files/20052616/452381-console.zip)
  

The node that managed to TX across the links without being keyed up (‘unwanted TX key up’) had node number 1979 and IP address 192.168.1.29.  
From the tcpdump of node 452381 it looks like to me that they key up was maybe 60ms of audio, and the audio data looked like it was all hexadecimal 0xf (tcpdump time 1034.810074).

I was able to test these recent versions of ASL3 and did not encounter 'hang up storm':
3.3.0
3.4.0

These versions I was able to test and did have 'hang up storm':
3.4.1
3.4.3
3.4.5

I did not test versions 3.4.2 or 3.4.4, but imagine they would also have problems.

Something added with version 3.4.1 seems to be the cause of ‘hangup storms’ and ‘unwanted TX key up’, and seems to affect HamVOIP nodes.

Matt

mkmer, 2025-05-06:
Is this error message :
```
[May 3 14:58:16] ERROR[464] app_rpt.c: No link messages from [452381] in 46 seconds -- One way audio??? Forcing reconnect.
[May 3 14:58:16] WARNING[464] app_rpt.c: Node 1998: Reconnect Attempt to 452381 in process
```
On the "non" ASL3.4.x node?

mpannen1979, 2025-05-06:
I have lined up and compared the tcpdumps from the 1998 node end and the 452381 node end, and I nothing seems to be missing on the wire for network packets.

Console times and tcpdump times line up like so:

wireshark filter:
_ws.col.info ~ "HANGUP" && ip.addr==192.168.1.10

452381 console hangup time (for node 1998): 14:58:16.384
452381 tcpdump time: 1024.384141

wireshark filter:
_ws.col.info ~ "HANGUP"

1998 Console hangup time: 14:58:16
1998 tcpdump time: 1608.290664

mkmer, 2025-05-06:
This is at least one "problem" found on the ALS3 side:
```[2025-05-03 12:34:56.675] WARNING[906][C-0000002c] app_rpt/rpt_channel.c: Failed to send text !NEWKEY1! on IAX2/192.168.1.10:4569-6749```

It's an indication that we can't send text messages on the IAX link (for some reason) -> which aligns with the HAMVOIP nodes complaining about not receiving link messages.

At one point there was a bug at the Asterisk level where text messages would become stuck if jitter buffer was enabled.  I'm not sure why 3.4.0 would be "ok" and 3.4.1 is "not ok" at the moment or if it has anything to do with JB.

Do you have a console log of the "good" behavior?  That would help us isolate what is different between the log files (maybe).

mpannen1979, 2025-05-06:
Most of the console log is good behavior, except for when it is not lol.

Was there some activity to capture that would be helpful?  Should I use a version of ASL3 that I know works nicely, or 3.4.1+?

The log entry you reference above would have been shortly after an 'astreset', if I recall, so 12:34:56 should have been about the time the nodes were trying to reconnect their permanent connections.

mkmer, 2025-05-06:
It IS bad if we don't get the `NEWKEY` message as it changes the behavior for keying - this would/could leave things keyed up that should not be.

First thing that happens on an IAX connection -> send `NEWKEY`, then expect a `NEWKEY` back, if we don't get it, we change to a "old key" mode (It appears to be some sort of legacy thing that we don't fully know how legacy it is).

The other entry I don't see on my nodes, and is a bit interesting:
```
[2025-05-03 14:40:55.930] DEBUG[892]: pbx_variables.c:819 pbx_substitute_variables_helper_full_location: Function IAXPEER(CURRENTCHANNEL) result is '192.168.1.11'
[2025-05-03 14:40:55.930] DEBUG[892]: pbx_variables.c:819 pbx_substitute_variables_helper_full_location: Function IAXPEER(CURRENTCHANNEL) result is '192.168.1.17'
[2025-05-03 14:40:55.931] DEBUG[892]: pbx_variables.c:819 pbx_substitute_variables_helper_full_location: Function IAXPEER(CURRENTCHANNEL) result is '192.168.1.28'
[2025-05-03 14:40:55.931] DEBUG[892]: pbx_variables.c:819 pbx_substitute_variables_helper_full_location: Function IAXPEER(CURRENTCHANNEL) result is '192.168.1.14'
```
Are these "common" in the older versions of ASL3 too? (aka I don't need to dig into it for this problem).

mpannen1979, 2025-05-06:
I will gather console logs from 3.3.0.  I will let it run for 4 hours or so, how's that?

mpannen1979, 2025-05-06:
No trouble with 3.3.0 or 3.4.0.  

I have started a console dump from 3.3.0 which includes as close to asterisk restart as I could from downgrade to 3.3.0.

mkmer, 2025-05-06:
@mpannen1979 I think we are onto something, don't worry about more data (unless you enjoy that stuff :) )
We actually may have addressed it in the current dev branch.  Just a bit more research and we will follow up.

mpannen1979, 2025-05-06:
OK nice

Here is the logs from 3.3.0 just because lol.

No hang up storms or problems as expected for 3.3.0

[452381-3-3-0.zip](https://github.com/user-attachments/files/20068559/452381-3-3-0.zip)
[1998-3-3-0.zip](https://github.com/user-attachments/files/20068558/1998-3-3-0.zip)

mkmer, 2025-05-22:
These PRs #636 , #632, #644 are the "fix".

closed 2025-05-22 by mkmer

mpannen1979, 2025-05-22:
I guess, is there a specific protocol requirement, or is HamVOIP just being picky.  Evetually things will work themselves out with ASL3 adoption I hope.

mkmer, 2025-05-25:
Hamvoip is trying to detect a mode problem by timing status text messages. Now that we fixed the timing issue, hamvoip should be fine. Just wanted to document what we "observed" for future investigations.