AllStarLink/app_rpt #681: ASL3 remote base not working

wb2dyb, 2025-05-13:
Trying to make the built in RB work. But it doesn't. Included in node rpt.conf I added:

remote = ft100
ioport = /dev/ttyACM0
iospeed = 4800

I know these configs are correct as I used them in ASL2 remote and also the port and speed work fine on the same PC with hamlib. 

If I change the nodes node line to delete the "Y" at the end of the line and then comment out the remote specific lines notes above this node 2428 can be connected to from another. But when remote base is enabled you cannot connect to the remote node. 

It was suggested to me to open a ticket because one person on the developer list is already running a remote. Maybe she or he can give me pointers

GeorgeC W2DB

PS just to make more certain I;m doing things correctly I swapped drives to a ASL2 installation. Remote worked fine.

wb2dyb, 2025-05-14:
I need to add one thing that may help find the problem. If I comment out the remote= line from rpt.conf I can connect to the node. Put the line back in and the node will not accept any connection. I swapped drives (uses a 16GB SSD) to one with ASL2 with remote base on and it works perfectly. Something with remote= is inhibiting connection. Thanks
GeorgeC W2DB

Allan-N, 2025-05-14:
Would you be OK sharing (attaching) the `rpt.conf` file from the system you are setting up as a remote node?

Can you also share the `rpt.conf` file from the system you are trying to connect from?

Lastly, have you looked at the output from `sudo asterisk -r -vvvv` from both systems when you are unsuccessful connecting?  Please share any interesting logs.

wb2dyb, 2025-05-15:
First I'll show an error that, when not being able to connect, shows up on the "remote base" computer in asterisk -r -vvvv mode:
**Connected to Asterisk 22.2.0+asl3-3.4.5-1.deb12 currently running on debian12-ASL (pid = 638)
    -- Accepting UNAUTHENTICATED call from 192.168.1.225:4570:
    --        > requested format = gsm,
    --        > requested prefs = (gsm|ulaw|adpcm|g726aal2|ilbc|speex16),
    --        > actual format = ulaw,
    --        > host prefs = (ulaw|adpcm|gsm),
    --        > priority = mine
    -- Executing [2428@radio-secure:1] NoOp("IAX2/192.168.1.225:4570-9414", "Connect from node: 2360") in new stack
    -- Executing [2428@radio-secure:2] NoOp("IAX2/192.168.1.225:4570-9414", "Connect to: 2428") in new stack
    -- Executing [2428@radio-secure:3] NoOp("IAX2/192.168.1.225:4570-9414", "The Channel IP is 192.168.1.225") in new stack
    -- Executing [2428@radio-secure:4] GotoIf("IAX2/192.168.1.225:4570-9414", "0?connect") in new stack
    -- Executing [2428@radio-secure:5] GotoIf("IAX2/192.168.1.225:4570-9414", "1?denylist") in new stack
    -- Goto (radio-secure,2428,9)
    -- Executing [2428@radio-secure:9] GotoIf("IAX2/192.168.1.225:4570-9414", "0?:connect") in new stack
    -- Goto (radio-secure,2428,12)
    -- Executing [2428@radio-secure:12] Rpt("IAX2/192.168.1.225:4570-9414", "2428") in new stack
[2025-05-14 19:14:12.707] WARNING[3652][C-0000000f]: app_rpt.c:6134 rpt_exec: Node 2428 is not ready yet, rejecting call on IAX2/192.168.1.225:4570-9414
  == Spawn extension (radio-secure, 2428, 12) exited non-zero on 'IAX2/192.168.1.225:4570-9414'
    -- Hungup 'IAX2/192.168.1.225:4570-9414'
debian12-ASL*CLI>**

_**********NEXT I COMMENT OUT THE remote= LINE:_

**Connected to Asterisk 22.2.0+asl3-3.4.5-1.deb12 currently running on debian12-ASL (pid = 3799)
    -- Accepting UNAUTHENTICATED call from 192.168.1.225:4570:
    --        > requested format = gsm,
    --        > requested prefs = (gsm|ulaw|adpcm|g726aal2|ilbc|speex16),
    --        > actual format = ulaw,
    --        > host prefs = (ulaw|adpcm|gsm),
    --        > priority = mine
    -- Executing [2428@radio-secure:1] NoOp("IAX2/192.168.1.225:4570-14125", "Connect from node: 2360") in new stack
    -- Executing [2428@radio-secure:2] NoOp("IAX2/192.168.1.225:4570-14125", "Connect to: 2428") in new stack
    -- Executing [2428@radio-secure:3] NoOp("IAX2/192.168.1.225:4570-14125", "The Channel IP is 192.168.1.225") in new stack
    -- Executing [2428@radio-secure:4] GotoIf("IAX2/192.168.1.225:4570-14125", "0?connect") in new stack
    -- Executing [2428@radio-secure:5] GotoIf("IAX2/192.168.1.225:4570-14125", "1?denylist") in new stack
    -- Goto (radio-secure,2428,9)
    -- Executing [2428@radio-secure:9] GotoIf("IAX2/192.168.1.225:4570-14125", "0?:connect") in new stack
    -- Goto (radio-secure,2428,12)
    -- Executing [2428@radio-secure:12] Rpt("IAX2/192.168.1.225:4570-14125", "2428") in new stack
[2025-05-14 19:17:59.399] ERROR[3901][C-00000009]: netsock2.c:303 ast_sockaddr_resolve: getaddrinfo("none", "(null)", ...): Name or service not known
[2025-05-14 19:17:59.399] WARNING[3901][C-00000009]: app_rpt.c:6002 parse_caller: Reported node 2360, name none cannot be found!!
  == Spawn extension (radio-secure, 2428, 12) exited non-zero on 'IAX2/192.168.1.225:4570-14125'
    -- Hungup 'IAX2/192.168.1.225:4570-14125'
debian12-ASL*CLI>**

Now in the second swet the 2360 node still cannot connect to 2428 
The difference between the two responses is the elimination of the "Y" character in the nodes line in the lower example plus that also has remote= commented out. In the upper example the 2428 node line ends in ",Y" and remote=ft100 is included in rpt.conf. 

I appreciate the help in getting ASL remote base working again.
GeorgeC W2DB
Crowley, TX

Attached is the rpt.conf file.

wb2dyb, 2025-05-15:
forgot the rpt.conf from the "connect from" node here it is as well as the 2428 node (remote base) that didn't seem to attach before

[rpt.conf.2428.txt](https://github.com/user-attachments/files/20221861/rpt.conf.2428.txt)

[rpt.conf.2360.txt](https://github.com/user-attachments/files/20221736/rpt.conf.2360.txt)

mkmer, 2025-05-15:
Sounds like it's related to this PR
https://github.com/AllStarLink/app_rpt/pull/532/

Maybe not setting `myrpt->ready` for remote configuration....

closed 2025-05-16 by mkmer

wb2dyb, 2025-05-16:
This is all new to me. Does this mean that if I do use apt I can get the updated asterisk executable?GeorgeC Sent from my iPhone

Closed #681 as completed via #686.

Allan-N, 2025-05-16:
>This is all new to me. Does this mean that if I do use apt I can get the updated asterisk executable?

Almost.  This means that we have "a" fix (that we believe should take care of the issue).  The change has been merged into the `app_rpt` source repository but it has not yet been built/released in an `apt` package.  We should have a "beta" release available towards the end of this month.

Let us know if you are itching to try out the fix, are OK with being on the bleeding edge, are very comfortable with the Linux CLI, and want to do some early testing.

wb2dyb, 2025-05-16:
Very familiar with Linux client, started with Slackware in 1994I’d love to try it but don’t go thru any extra work in account of me but really hoping remote base will work. I used it back to 2010 when Jim duuuude fixed the yaesu serial command timing for me. GeorgeC Sent from my iPhone

PR #686 (2025-05-16): app_rpt: rpt_vars[]->ready never set true for remote.  Issue created by PR#532
Resolves #681