AllStarLink/app_rpt #684: chan_tlb: unable establish connection in latest update 22.2.0+asl3-3.4.5

k6ian, 2025-05-14:
Prior to update chan_tlb worked and linked up. Post update to 22.2.0+asl3-3.4.5, connection for chan_tlb wouldn’t link. Rolled back to previous version and connection ability restored. What type of log file or data can I provide to assist with troubleshooting?

mkmer, 2025-05-14:
Any log files, details related to the "failure" would be VERY helpful.

mkmer, 2025-05-14:
Couple related "fixes".  
https://github.com/AllStarLink/app_rpt/pull/685
https://github.com/AllStarLink/app_rpt/commit/e7eb9371b8f45e415b1c886dcf8b624cebaeec97

@k6ian : Can you get a log from the TLB box - see if it's complaining about incorrect RTP type.

Allan-N, 2025-05-14:
Also, https://github.com/AllStarLink/app_rpt/pull/682

closed 2025-05-27 by Allan-N