AllStarLink/app_rpt #695: Support Asterisk AGC() Automatic Gain Control function

davidgsd, 2025-06-03:
Asterisk has an AGC() dialplan function - https://docs.asterisk.org/Asterisk_22_Documentation/API_Documentation/Dialplan_Functions/AGC/ - that could potentially be very useful for hub/repeater nodes that have a lot of users with widely varying audio levels, as well as for personal nodes who might connect to other such nodes that do not have any AGC function enabled.

To use the AGC function it appears it is necessary to (a) recompile Asterisk to enable the libspeex module, and to (b) figure out and document where in extensions.conf the AGC() function needs to be called in order to implement the function on incoming and/or outgoing connections.

Theoretically this should be very easy to do for one of the ASL devs who has familiarity with the app_rpt dialplans and who can add the libspeex option to the asl3-asterisk build settings. And it should be very low risk as it would only enable a small module and have no effect until an AGC() call was manually added by users in extensions.conf.

jxmx, 2025-06-03:
The ASL3 packages for Debian 12 are already compiled with libspeex1 and libspeexdsp1. No code/rebuilding for that to be present is required.

```
jdm@node460181:/usr/lib/aarch64-linux-gnu/asterisk/modules $ ldd * | grep speex
codec_speex.so:
        libspeex.so.1 => /lib/aarch64-linux-gnu/libspeex.so.1 (0x0000007fae600000)
        libspeexdsp.so.1 => /lib/aarch64-linux-gnu/libspeexdsp.so.1 (0x0000007fae5d0000)
format_ogg_speex.so:
        libspeex.so.1 => /lib/aarch64-linux-gnu/libspeex.so.1 (0x0000007fa1df0000)
func_speex.so:
        libspeexdsp.so.1 => /lib/aarch64-linux-gnu/libspeexdsp.so.1 (0x0000007f87c90000)
```

Allan-N, 2025-06-03:
Minimally, it looks like you will need add the following to your `modules.conf` file
```
load = func_speex.so
```

davidgsd, 2025-06-03:
Does anyone have any which contexts in extensions.conf the AGC() calls should be added to?  From the API documentation,
```
Example: Apply automatic gain control
exten => 1,1,Set(AGC(rx)=8000)
exten => 1,2,Set(AGC(tx)=off)
```
What context in extensions.conf applies to outgoing connections? And which of them would be best for incoming connections from other nodes?

davidgsd, 2025-06-03:
In iax.conf the radio context is listed under the comment `; Incoming radio connections`, and there doesn't seem to be any similar comment about outgoing connections. There is an allstar-sys context that seems like it might be doing something relating to outgoing connections? That context in extensions.conf is:
```
[allstar-sys]
exten => _1.,1,Rpt(${EXTEN:1}|Rrpt/node:NODE:rpt/in-call:digits/0:PARKED|120)
exten => _1.,n,Hangup

exten => _2.,1,Ringing
exten => _2.,n,Wait(3)
exten => _2.,n,Answer
exten => _2.,n,Playback(rpt/node)
exten => _2.,n,Saydigits(${EXTEN:1})
exten => _2.,n,Rpt(${EXTEN:1}|P|${CALLERID(name)}-P)
exten => _2.,n,Hangup
```
... above block repeats _3, _4, _5 ...

Is that first line with call to Rpt() what enables app-rpt to initiate a call to connect to a node?  If so maybe the following likely to work?
[allstar-sys]
**exten => _1.,1,Set(AGC(rx)=8000)
exten => _1.,n,Set(AGC(tx)=off)**
exten => _1.,n,Rpt(${EXTEN:1}|Rrpt/node:NODE:rpt/in-call:digits/0:PARKED|120)
exten => _1.,n,Hangup
...

jxmx, 2025-06-03:
`[allstar-sys]` is for incoming Phone and Web Portal connections. You could set it there as well in the same way where you show it.

davidgsd, 2025-08-10:
Hello ASL devs,  bumping this issue in the hopes of getting an answer to my above question of how to enable the Asterisk AGC function. Please see the previous comment to see the suggestion I tried, which seemed to have no effect. How can I enable further debug and figure out how to enable this module? Thank you for any tips on how to make this work.

There is also another reason why this feature could be very useful: C-Media ICs have a max output level of around 3Vpp depending on the specific version (108/119/A/B), which should be sufficient for radios with flat audio in (eg. Kenwood TM-V71A which has an input voltage spec of 2Vpp), and is sufficient if in fact the incoming IAX audio peak levels are >= ~-3dBFS. However in reality peak IAX audio levels have significant variation, between a range of -10 to -3 dBFS peak based on some quick tests I did on a few repeater nodes. If a user happens to mainly use a system with levels closer to -10dBFS, the audio going out the URI will be significantly lower then the 2Vpp required by many radios. This could be addressed by adding an OpAmp in the URI, or by adding an configurable ~3-6dB Tx boost (with limiting) in USBRadio, but if the AGC module works well that may be a more ideal solution.

davidgsd, 2025-08-12:
> Assuming your console verbosity is at least 3, you should see all the dialplan that is executed, no debug is needed.
> 
> e.g. `core set verbose 10`

I still see no messages indicating anything about any dialplan actions or my calls added to AGC(). 

Here is what I have in extensions.conf.  I am running very latest ASL with all .conf files fully up-to-date.

...

[radio-secure]
;exten => ${NODE},1,rpt(${EXTEN})
exten => _XXXX!,1,Set(NODENUM=${CALLERID(num)})
        same => n,NoOp(Connect from node: ${NODENUM})
        same => n,NoOp(Connect to: ${EXTEN})
        same => n,NoOp(Channel type: ${CHANNEL(channeltype)})
        same => n,GotoIf($["${CHANNEL(channeltype)}" = "IAX2"]?:allowlist)              ; if not IAX2, skip IP check

        **; Lines added to enable AGC
        same => n,NoOp(Enable AGC radio-secure)
        same => n,Set(AGC(rx)=12000)
        same => n,Set(AGC(tx)=12000)**

        ; if peer IP is localhost, connect
        same => n,NoOp(Channel Peer IP: ${CHANNEL(peerip)})
        same => n,GotoIf($["${CHANNEL(peerip)}" = "127.0.0.1"]?connect)                 ; if localhost; connect

        ; if allowlist/extension doesn't exist, check the denylist
        same => n(allowlist),GotoIf($[${DB_KEYCOUNT(allowlist/${EXTEN})} = 0]?denylist) ; no allowlist, check denylist
        ; if allowlist/extension/callerID exists, connect
        same => n,GotoIf(${DB_EXISTS(allowlist/${EXTEN}/${NODENUM})}?connect)           ; in allowlist, connect
        same => n,NoOp(${EXTEN} not in allowlist, Hangup)
        same => n,Hangup

        ; if denylist/extension/callerID doesn't exist, connect
        same => n(denylist),GotoIf(${DB_EXISTS(denylist/${EXTEN}/${NODENUM})}?:connect) ; not in denylist, connect
        same => n,NoOp(${EXTEN} is in denylist, Hangup)
        same => n,Hangup

        ; connect!
        same => n(connect),rpt(${EXTEN})
        same => n,Hangup

...

[allstar-sys]
exten => _1.,1,Rpt(${EXTEN:1},Rrpt/node:NODE:rpt/in-call:digits/0:PARKED,120)

**; Lines added to enable AGC
exten => _1.,n,NoOp(Enable AGC allstar-sys)
exten => _1.,n,Set(AGC(rx)=12000)
exten => _1.,n,Set(AGC(tx)=12000)**

exten => _1.,n,Hangup

exten => _2.,1,Ringing
exten => _2.,n,Wait(3)
exten => _2.,n,Answer
exten => _2.,n,Playback(rpt/node)
exten => _2.,n,Saydigits(${EXTEN:1})
exten => _2.,n,Rpt(${EXTEN:1},P,${CALLERID(name)}-P)
exten => _2.,n,Hangup

exten => _3.,1,Ringing
exten => _3.,n,Wait(3)
exten => _3.,n,Answer
exten => _3.,n,Playback(rpt/node)
exten => _3.,n,Saydigits(${EXTEN:1})
exten => _3.,n,Rpt(${EXTEN:1},Pv,${CALLERID(name)}-P)
exten => _3.,n,Hangup

exten => _4.,1,Ringing
exten => _4.,n,Wait(3)
exten => _4.,n,Answer
exten => _4.,n,Playback(rpt/node)
exten => _4.,n,Saydigits(${EXTEN:1})
exten => _4.,n,Rpt(${EXTEN:1},D,${CALLERID(name)}-P)
exten => _4.,n,Hangup

exten => _5.,1,Ringing
exten => _5.,n,Wait(3)
exten => _5.,n,Answer
exten => _5.,n,Playback(rpt/node)
exten => _5.,n,Saydigits(${EXTEN:1})
exten => _5.,n,Rpt(${EXTEN:1},Dv,${CALLERID(name)}-P)
exten => _5.,n,Hangup

I then run asterisk -vvvr and then when I connect I do not see anything relating the above eg. the NoOp functions which should be logging a message??? And I hear no difference in audio levels. 

Below I connect to another node and it shows only that the call was accepted, nothing about any dialplan functions.

node571570*CLI> core set verbose 10
Console verbose was 3 and is now 10.
  == Manager 'admin' logged on from 127.0.0.1
  == Manager 'admin' logged off from 127.0.0.1
    -- Call accepted by 205.178.34.59:4572 (format g726aal2)
    -- Format for call is (g726aal2)
node571570*CLI>

Does anyone have any ideas how I can enable AGC() and actually see log messages relating to the dial plan?  Should I maybe ask on the Asterisk forum?

InterLinked1, 2025-08-12:
Strange, it seems like it should work... Have you reloaded the logger? (`logger reload`)
Do you ever see verbose messages in your console? Typically this is purple/blue colored output.

davidgsd, 2025-08-12:
Doing a logger reload gave a permissions error and turns out /var/log/asterisk/messages.log was owned by root with 640 permissions, so I chown'd everything in /var/log/asterisk/ to asterisk:asterisk and restarted Asterisk, and now logger reload does not return an error but I still see no verbose messages.

```
node571570*CLI> core set verbose 10
Console verbose is still 10.
node571570*CLI> logger reload
 Asterisk Queue Logger restarted
node571570*CLI> core set verbose 10
Console verbose is still 10.
  == Manager 'admin' logged on from 127.0.0.1
  == Manager 'admin' logged off from 127.0.0.1
    -- Call accepted by 44.15.69.230:4568 (format g726aal2)
    -- Format for call is (g726aal2)
```

InterLinked1, 2025-08-12:
> Doing a logger reload gave a permissions error and turns out /var/log/asterisk/messages.log was owned by root with 640 permissions, so I chown'd everything in /var/log/asterisk/ to asterisk:asterisk [...]

What happens if you add:

``full.txt => notice,warning,error,dtmf,debug,verbose``

Do you see them in the new full.txt log file?

davidgsd, 2025-08-12:
I just tried all the above on 3 different nodes, that were built at completely different times, and have different configurations (radio-less / HT node / USBRadio node), and they all do the exact same thing.  These are plain vanilla ASL3 installs that I have not made any significant or unusual changes to.

Therefore I would have to conclude that this issue is definitely _not_ something specific to "my" system, and that ASL3 simply does not show dialplan messages in the CLI.  I will open a separate bug report about this.

Back to this issue - which is getting a simple dialplan function to actually work, would you or another ASL dev (who presumably somehow is actually able to see dialplan log messages) be willing to try calling the AGC() function as I have described above and see if you can get it to work? Thank you.

davidgsd, 2025-08-13:
According to https://allstarlink.github.io/config/extensions_conf/ allstar-sys stanza is for the Telephone Portal (reverse autopatch), and there is no mention of any context being for outbound connections. And a web search indicates that Asterisk is supposed to  have context(s) for outbound calls defined in extensions.conf.  So perhaps the issue here is that there was an oversight in the app-rpt design where it does not provide a way to call a extensions.conf context when making outbound IAX connections?

davidgsd, 2025-08-13:
Not sure what you mean by change, nothing changed. In my earlier testing I was testing outbound connects, and as stated, those were not executing any context in extensions.conf and thus no associated log messages would be printed.

So back to the question - **how can the AGC() function be executed when making outbound IAX connections**?

InterLinked1, 2025-08-13:
> Not sure what you mean by change, nothing changed. In my earlier testing I was testing outbound connects, and as stated, those were not executing any context in extensions.conf and thus no associated log messages would be printed.
> 
> So back to the question - how can ASL be set up to call the AGC() dialplan function when making outbound IAX connections?

In general, you can run dialplan on the callee side of the created channel by using a pre-dial handler. See https://docs.asterisk.org/Configuration/Dialplan/Subroutines/Pre-Dial-Handlers/#overview

davidgsd, 2025-08-13:
Thanks for the reply, but unfortunately the page you linked is not at all clear on how I would actually do this in ASL.

It says that Pre-dial handlers are supported in the Dial and FollowMe applications, using a 'b' or 'B' option, but I thought the application here was Rpt - not Dial or FollowMe?

And in an example it refers to the [default] context, but the earlier link I posted states that's "the stanza Asterisk uses when a more specific stanza is not available or if a dialplan error occurs", and I already confirmed that stanza is not called when outbound connections are made, and you further clarified that contexts are not used for outbound calls. So this link you have posted seems to just go in circles.  If contexts are not for outbound calls, then where should I put the 'b'/'B' option? Apparently not in extensions.conf, so where then should it go? Does app-rpt actually call Dial() somewhere in one of the other config files?

All I'm trying to do here is make a simple call to a simple function, AGC().

It would be great if someone could just tell me what specific change I need to make in which specific file so that AGC() will be called when I make an outbound connection.

This could be a very useful feature for many users and in a variety of use cases, and fundamentally it seems that app-rpt should be able to call a dialplan function when making an outbound connection, and that it shouldn't require weeks of going back and forth in a thread like this just to figure out how to make that happen.  Thanks in advance to anyone who can figure out how to actually make this happen.