AllStarLink/app_rpt #700: Audio from Echolink Choppy

w0pd, 2025-06-16:
I have choppy Echolink receive audio coming from Echolink to my transmitter. I think it all started with the latest update as it was fine prior to that. The Allstar audio is great both TX and RX. but audio coming from Echolink is choppy. Reports that my audio going out to Echolink is just fine.  I have tried to lower the erxgain amd can make the audio level lower, but the choppy sound continues.  I have rebooted many times and ensured the software is the latest. According to the discusion boards, a few other people noticed this behavior as well. 

I am running:
Raspberry Pi 5 Model B Rev 1.0
Debian GNU/Linux 12 (Bookworm)
OS Kernal 6.12.25+rpt-rpi-2712
Asterisk 22.2.0+asl3.4.5-1.deb12
ASL 3.4.5
allmon3 1.4.2-1.deb12
asl3 3.7.1-1.deb
URIxA 9095 With internal EEPROM using a CM119A audio controller chip 
Connected to a Motorola MTR3000

I have pasted the data from the following files here:
1. etc/asterisk/echolink.conf
2. etc/asterisk/rpt.conf
3. etc/asterisk/usbradio.conf

Thanks
Mike W0PD
w0pd@outlook.com
**------------------------------------------------------------/etc/asterisk/echolink.conf---------------**

```
; If you are going to use this channel driver, you MUST enable it in modules.conf
; change:
; noload => chan_echolink.so 		; echolink Channel Driver
; to:
; load => chan_echolink 			; echolink Channel Driver

[el0]
call = W0PD-R						; Change this!
pwd = hidden					; Change this!
name = Mike McConnell					; Change this!
qth = Wellsburg, Iowa					; Change this!
email = w0pd@outlook.com				; Change this!
node = 218084			                        ; Change this!
; Data for EchoLink Status Page
lat = 42.430770							; Latitude in decimal degrees
lon = 92.927372						; Longitude in decimal degrees
freq = 442.325                          ; not mandatory Frequency in MHz
tone = 136.5                          ; not mandatory CTCSS Tone (0 for none)
power = 50							; 0=0W, 1=1W, 2=4W, 3=9W, 4=16W, 5=25W, 6=36W, 7=49W, 8=64W, 9=81W (Power in Watts)
height = 60							; 0=10 1=20 2=40 3=80 4=160 5=320 6=640 7=1280 8=2560 9=5120 (AMSL in Feet)
gain = 5							; Gain in db (0-9)
dir = 0								; 0=omni 1=45deg 2=90deg 3=135deg 4=180deg 5=225deg 6=270deg 7=315deg 8=360deg (Direction)

maxstns = 20						; Max Stations

rtcptimeout = 10					; Max number of missed heartbeats from EL
recfile = /tmp/echolink_recorded.gsm	;
astnode = 568190						; Change this!
context = radio-secure				; Default in code is echolink-in

; Max 4 servers
server1 = nasouth.echolink.org
server2 = naeast.echolink.org
server3 = servers.echolink.org
server4 = backup.echolink.org

; To deny w6xxx you would add the statement: deny = w6xxx
; To prohibit computer-based connections you would write: permit = *-*
; To allow access to only a select group of callsigns: permit = w6abc,w6def,...

; permit							; comma delimited list of callsign, type (-r)
; deny

; Remote text commands thru netcat:
; o.conip <IPaddress>    (request a connect)
; o.dconip <IPaddress>   (request a disconnect)
[... 865 lines cut ...]
; Note: the device string is automatically found when the
;       USB setting "devstr=" is empty.
;
; Note: the interface "tune" settings will be added to the
;       per-node settings (below).

[568190](node-main)
rxdemod = flat
rxboost = no

carrierfrom = usbinvert
ctcssfrom = no

;;;;; ASL3 Tune settings ;;;;;
devstr = 1-1:1.0
rxmixerset = 999
txmixaset = 650  ;was 300
txmixbset = 400
rxvoiceadj = 1.072188
rxctcssadj = 4.565804
txctcssadj = 0
rxsquelchadj = 0
fever = 0
rxondelay = 0
```

Allan-N, 2025-06-16:
Issue moved to app_rpt.

Q's.
- are you having issues with specific EL nodes?  or is it the audio from all connected EL nodes?
- do you know anything about the problematic EL nodes?  For example, are the connected nodes running on an EchoLink server?  or an ASL server w/chan_echolink?
- how are you connected with the EL nodes?  Are these connections that you established?  or nodes that have connected with you?
- can you characterize the EL audio level (the received audio is quiet, normal, loud)?

jxmx, 2025-06-16:
> Both chan_simpleusb and chan_usbradio show the problem per N2DYI.

Can you clarify what you mean by this? Are you saying that apart from Echolink, the USB device channels have this problem SEPARATELY? Or just that you can hear the issue when the audio is played.

okcrum, 2025-06-16:
From N2DYI:
"I tried with both simpleusb and usbradio. The result is consistent. For whatever reason, audio is always choppy from Echolink to any USB device. I tried with an Aursinc, a real SHARI and a UCI120. I haven't tried piping the audio of app_rpt directly to the codec on the Raspberry Pi using outstreamcmd and aplay yet, though. It seems as if other connected ASL nodes also receive choppy audio from Echolink until the sound fob is disassociated with the node."

Allan-N, 2025-06-17:
Looking at the choppy audio this morning and found something interesting.  I enabled `archiveaudio` on one of my systems, connected to an EL node, and listened to some of the traffic.  Yes, the audio didn't sound great.  Then, I went back and listened to the captured (archived) audio ... that I believe sounded much better.

Was this just because I was listening to the captured audio on my desktop with much better speakers?  or is this pointing to an issue with the audio path between `chan_echolink`, `app_rpt`, and `chan_simpleusb`?

wd5m, 2025-06-25:
This issue may also be affecting chan_tlb audio.  I was going to capture the audio with the archive feature, but It looks like chan_tlb audio is not saved with archive. 

David, WD5M

wd5m, 2025-06-28:
I seem to be hearing this audio distortion from iaxclient connections too. I have archivedir recordings, if needed. It does not seem to happen as often as with chan_echolink or chan_tlb. Maybe hours between occurrences.  This is with the latest ASL3 release on a VPS host. 

David, WD5M

fjmartin1966, 2025-07-02:
I wanted to echo the comment I had put on the ASL Forum and add some additional information:

Original note:
I wanted to add that I built a new system in the past week that is experiencing this choppy audio issue also. I then spoke with a buddy who has a bunch of different nodes for various jobs and he also has a HamVoIP installation. A few of us used both our PCs and Phones to make EchoLink calls to newer ASL3 nodes and experienced the rhythmic choppy audio coming from an EchoLink client. Audio to the client passed through the ASL3 node did not experience this. We then repeated the same test calls but to his old HamVoIP node and experienced excellent quality audio in both directions.
My configuration:
ASL3 3.018 appliance build and then any updates using apt. EchoLink enabled on ASL3. Also have AllScan 0.93 and DVSwitch latest update and being used strictly for DMR at this point
Hardware is Pi 5 4GB. 64GB Class 10 Micro SD. SR-FRS UHF radio/sound card
Internal Private node built to bridge DVSwitch to AllStar primary node

Update:
I disabled EchoLink on my ASL3 node and built up a separate Pi with HamVoIP as a radioless private node with EchoLink configured on it.  I then connect this HamVoIP private node to my ASL3 primary node and the audio is perfect.  No issues as seen on ASL3 with EchoLink.  I have also tested with users connected/bridged via AllStar, EchoLink and DMR (through DVSwitch).  Perfect audio.

Joe - KK7PZE

mkmer, 2025-07-02:
Because we have a local HUB that hosts nets using echolink and are not experiencing this issue, I'm looking to clarify a few things:

- Is the choppy audio ONLY on a local USB sound card?
- Is the choppy audio "forwarded" onto any connected IAX clients?
- When you capture the audio, what method is being used?

If this is a usb sound card, with debug set to 7 or higher, are you seeing any of these errors (there should be 1 for each start of transmission):
 `"A null frame has been added"` ?

wd5m, 2025-07-02:
> Because we have a local HUB that hosts nets using echolink and are not experiencing this issue, I'm looking to clarify a few things:

>     * Is the choppy audio ONLY on a local USB sound card?

I experience choppy audio on a VPS HUB node connected to the EchoLink conference node *SPACE*.  No USB sound card is attached to the hub. 

>     * Is the choppy audio "forwarded" onto any connected IAX clients?

Other connected nodes are all Allstar systems connected through rpt via [radio-secure].

>     * When you capture the audio, what method is being used?

I use the archivedir= feature of rpt.conf to record audio on the HUB node. 

[20250702160745.wav](https://space.rfnet.link/recordings/20250702160745.wav)
[20250702160751.wav](https://space.rfnet.link/recordings/20250702160751.wav)
[20250702160801.wav](https://space.rfnet.link/recordings/20250702160801.wav)
[20250702160805.wav](https://space.rfnet.link/recordings/20250702160805.wav)

David, WD5M

jbdubbs, 2025-07-10:
This problem is also present as above described on node 27664.  Please message me if you'd like diagnostics or configs from our Rpi 4.

OS            : Debian GNU/Linux 12 (bookworm)                                                     
                       │ OS Kernel     : 6.12.25+rpt-rpi-v8                                                                 
                       │                                                                                                    
                       │ Asterisk      : 22.4.1+asl3-3.5.2-1.deb12                                                          
                       │ ASL [app_rpt] : 3.5.2                                                                              
                       │                                                                                                    
                       │ Installed ASL packages :                                                                           
                       │                                                                                                    
                       │   Package                         Version                                                          
                       │   ==============================  ==============================                                   
                       │   allmon3                         1.4.3-1.deb12                                                    
                       │   asl3                            3.8-1.deb                                                        
                       │   asl3-asterisk                   2:22.4.1+asl3-3.5.2-1.deb12                                      
                       │   asl3-asterisk-config            2:22.4.1+asl3-3.5.2-1.deb12                                      
                       │   asl3-asterisk-modules           2:22.4.1+asl3-3.5.2-1.deb12                                      
                       │   asl3-menu                       1.14-1.deb12

InterLinked1, 2025-07-14:
If it's still not clear what the offending change is, I would recommend doing a `git bisect`. That will very clearly point out the problematic change.

Some manual testing will be needed for each iteration, but it shouldn't take very long.

okcrum, 2025-07-15:
All inbound EL connections show the problem, whether one or many. I specifically tested multiple connections to make sure of that. Sorry for the fuzzy wording.

okcrum, 2025-07-19:
I just tested a patch from @Allan-N that fixes both the local VM and RPi 5 + USB hotspot test cases. Stand by to test.