AllStarLink/app_rpt #701: app_rpt.c:1010 perform_statpost returns html on failure

tsawyer, 2025-06-19:
Statpost failure prints html on Asterisk console. Is this intended?

```
********** AllStarLink [ASL] Version Info **********

OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-34-amd64

Asterisk      : 22.2.0+asl3-3.4.5-1.deb12
ASL [app_rpt] : 3.4.5

Installed ASL packages :

  Package                         Version
  ==============================  ==============================
  asl3                            3.7.1-1.deb
  asl3-asterisk                   2:22.2.0+asl3-3.4.5-1.deb12
  asl3-asterisk-config            2:22.2.0+asl3-3.4.5-1.deb12
  asl3-asterisk-doc               2:22.2.0+asl3-3.4.5-1.deb12
  asl3-asterisk-modules           2:22.2.0+asl3-3.4.5-1.deb12
  asl3-menu                       1.13-1.deb12
  dahdi                           1:3.1.0-2
  dahdi-dkms                      1:3.4.0-6+asl
  dahdi-linux                     1:3.4.0-6+asl
```

```
[2025-06-19 06:05:34.202] WARNING[1234175]: app_rpt.c:1010 perform_statpost: statpost to URL 'http://stats.allstarlink.org/uhandler?node=2521&time=1750338333&seqno=143535&nodes=T2530&apprptvers=3.4.5&apprptuptime=2388617&totalkerchunks=12&totalkeyups=2032&totaltxtime=29599&timeouts=0&totalexecdcommands=0' failed with code 503 : <!DOCTYPE html>
<html lang="en">
    <head>
        <meta charset="utf-8">
        <meta name="viewport" content="width=device-width, initial-scale=1">

        <title>Service Unavailable</title>

        <!-- Fonts -->
        <link rel="dns-prefetch" href="//fonts.gstatic.com">
        <link href="https://fonts.googleapis.com/css?family=Nunito" rel="stylesheet">

        <!-- Styles -->
        <style>
            html, body {
                background-color: #fff;
                color: #636b6f;
                font-family: 'Nunito', sans-serif;
                font-weight: 100;
                height: 100vh;
                margin: 0;
            }

            .full-height {
                height: 100vh;
            }

            .flex-center {
                align-items: center;
                display: flex;
                justify-content: center;
            }

            .position-ref {
                position: relative;
            }

            .code {
                border-right: 2px solid;
                font-size: 26px;
                padding: 0 15px 0 15px;
                text-align: center;
            }

            .message {
                font-size: 18px;
                text-align: center;
            }
        </style>
    </head>
    <body>
        <div class="flex-center position-ref full-height">
            <div class="code">
                503            </div>

            <div class="message" style="padding: 10px;">
                Service Unavailable            </div>
        </div>
    </body>
</html>
```

InterLinked1, 2025-06-19:
> I believe it was just supposed to copy the "reason" to the console (besides just `503`). Have to dig into what happened here.

I think setting `CURLOPT_FAILONERROR` would probably fix this

Allan-N, 2025-06-19:
There are clear diffs in how we handle curl errors in app_rpt.c and res_rpt_http_registrations.c ... and I like what I see in the latter.