AllStarLink/app_rpt #709: app_rpt 3.5.3 bug - rollback to 3.5.2 works fine

jxmx, 2025-06-30:
```
[2025-06-30 23:53:44.403] Asterisk 22.4.1+asl3-3.5.3-1.deb12 built by builder @ allstarlink.org on a aarch64 running
Linux on 2025-06-30 17:39:19 UTC
[2025-06-30 23:53:44.415] NOTICE[2064] loader.c: 69 modules will be loaded.
[2025-06-30 23:53:44.434] NOTICE[2064] cdr.c: CDR simple logging enabled.
[2025-06-30 23:53:44.435] NOTICE[2064] dnsmgr.c: Managed DNS entries will be refreshed every 300 seconds.
[2025-06-30 23:53:44.440] NOTICE[2064] indications.c: Default country for indication tones: us
[2025-06-30 23:53:44.440] NOTICE[2064] indications.c: Setting default indication country to 'us'
[2025-06-30 23:53:44.491] NOTICE[2064] res_smdi.c: No SMDI interfaces are available to listen on, not starting SMDI listener.
[2025-06-30 23:53:44.493] WARNING[2064] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2025-06-30 23:53:44.493] WARNING[2064] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2025-06-30 23:53:44.493] NOTICE[2064] chan_dahdi.c: Ignoring any changes to 'userbase' (on reload) at line 28.
[2025-06-30 23:53:44.493] NOTICE[2064] chan_dahdi.c: Ignoring any changes to 'vmsecret' (on reload) at line 36.
[2025-06-30 23:53:44.493] NOTICE[2064] chan_dahdi.c: Ignoring any changes to 'hasiax' (on reload) at line 40.
[2025-06-30 23:53:44.493] NOTICE[2064] chan_dahdi.c: Ignoring any changes to 'hasmanager' (on reload) at line 48.
[2025-06-30 23:53:44.493] WARNING[2064] chan_dahdi.c: Only FXO signalled channels may belong to a call group
[2025-06-30 23:53:44.493] WARNING[2064] chan_dahdi.c: Only FXO signalled channels may belong to a pickup group
[2025-06-30 23:53:44.554] WARNING[2064] loader.c: Module 'res_adsi' has been loaded but may be removed in a future release.
[2025-06-30 23:53:44.565] WARNING[2070] pbx_config.c: users.conf is deprecated and will be removed in a future version of Asterisk
[2025-06-30 23:53:44.756] NOTICE[2108] app_rpt.c: Domain used for DNS node lookup is: nodes.allstarlink.org
[2025-06-30 23:53:44.756] NOTICE[2108] app_rpt.c: Normal Repeater Init  48496
[2025-06-30 23:53:44.757] NOTICE[2108] app_rpt.c: Normal Repeater Init  430780
[2025-06-30 23:53:44.757] NOTICE[2108] app_rpt.c: Normal Repeater Init  430781
[2025-06-30 23:53:44.758] NOTICE[2108] app_rpt.c: Normal Repeater Init  430782
[2025-06-30 23:53:44.758] NOTICE[2108] app_rpt.c: Normal Repeater Init  430783
[2025-06-30 23:53:44.758] NOTICE[2108] app_rpt.c: Normal Repeater Init  430784
[2025-06-30 23:53:44.759] NOTICE[2108] app_rpt.c: Normal Repeater Init  430785
[2025-06-30 23:53:44.762] WARNING[2116] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-597581063
[2025-06-30 23:53:44.762] WARNING[2116] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.762] ERROR[2116] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.762] WARNING[2116] app_rpt.c: Disabled broken repeater 430781
[2025-06-30 23:53:44.772] WARNING[2120] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-1444085604
[2025-06-30 23:53:44.772] WARNING[2120] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.772] ERROR[2120] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.772] WARNING[2120] app_rpt.c: Disabled broken repeater 430785
[2025-06-30 23:53:44.772] WARNING[2115] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-2146888778
[2025-06-30 23:53:44.772] WARNING[2115] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.772] ERROR[2115] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.772] WARNING[2115] app_rpt.c: Disabled broken repeater 430780
[2025-06-30 23:53:44.772] WARNING[2117] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-1489462214
[2025-06-30 23:53:44.772] WARNING[2117] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.772] ERROR[2117] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.772] WARNING[2117] app_rpt.c: Disabled broken repeater 430782
[2025-06-30 23:53:44.773] WARNING[2118] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-1074182228
[2025-06-30 23:53:44.773] WARNING[2118] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.773] ERROR[2118] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.773] WARNING[2118] app_rpt.c: Disabled broken repeater 430783
[2025-06-30 23:53:44.773] WARNING[2119] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-149754002
[2025-06-30 23:53:44.773] WARNING[2119] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.773] ERROR[2119] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.773] WARNING[2119] app_rpt.c: Disabled broken repeater 430784
[2025-06-30 23:53:44.774] WARNING[2114] app_rpt/rpt_bridging.c: app_rpt/rpt_bridging.c:376 (dahdi_conf_add) Unable to set conference mode on DAHDI/pseudo-981632046
[2025-06-30 23:53:44.774] WARNING[2114] app_rpt/rpt_bridging.c: Failed to join DAHDI conf (mode: 7)
[2025-06-30 23:53:44.774] ERROR[2114] app_rpt/rpt_bridging.c: app_rpt.c:2788: Failed to add to conference using chan type 0
[2025-06-30 23:53:44.774] WARNING[2114] app_rpt.c: Disabled broken repeater 48496
[2025-06-30 23:53:45.590] WARNING[2097] chan_iax2.c: Silently dropping frame without existent call number: 7170
[2025-06-30 23:53:47.589] WARNING[2099] chan_iax2.c: Silently dropping frame without existent call number: 7170
[2025-06-30 23:53:47.632] WARNING[2149][C-0000001d] app_rpt.c: Cannot find specified system node 48496
[2025-06-30 23:53:51.496] WARNING[2150][C-0000001e] app_rpt.c: Cannot find specified system node 48496
[2025-06-30 23:53:52.752] WARNING[2151][C-0000001f] app_rpt.c: Cannot find specified system node 48496
[2025-06-30 23:53:53.966] WARNING[2105] chan_iax2.c: Silently dropping frame without existent call number: 470
[2025-06-30 23:53:55.966] WARNING[2098] chan_iax2.c: Silently dropping frame without existent call number: 470
[2025-06-30 23:53:57.407] WARNING[2152][C-00000020] app_rpt.c: Cannot find specified system node 48496
[2025-06-30 23:53:57.865] WARNING[2153][C-00000021] app_rpt.c: Cannot find specified system node 48496
[2025-06-30 23:53:59.019] WARNING[2154][C-00000022] app_rpt.c: Cannot find specified system node 48496
```

And so on. No config change from 3.5.2 to 3.5.3. Rolling the packages back to 3.5.2 results in a working system with, again, no config changes.

Allan-N, 2025-07-01:
FYI : a configuration with duplex=3 appears to be OK.

Allan-N, 2025-07-01:
Found it !

closed 2025-07-01 by Allan-N