AllStarLink/app_rpt #713: Unable to parse link string !IAXKEY! 1 1  after iaxrpt connection

ve2dje, 2025-07-01:
Saw  these errors/warnings in  CLI    when connect from iaxrpt. 

Not present in 3.3.0  saw the error in 3.5.2 and later  

Might not be related but I can not get  outstreamcmd=   to work

 -- Executing [470557@iaxrpt:1] Rpt("IAX2/iaxrpt-3182", "470557|X") in new stack
  == Spawn extension (iaxrpt, 470557, 1) exited non-zero on 'Surrogate/IAX2/iaxrpt-3182'
[2025-07-01 17:18:05.451] WARNING[8850]: app_rpt.c:1945 handle_link_data: Unable to parse link string !IAXKEY! 1 1

VE2DJE

Allan-N, 2025-07-01:
What do you know about the client / application that is connecting to your node?

Allan-N, 2025-07-01:
In [PR #588 ](https://github.com/AllStarLink/app_rpt/pull/588), we yanked what we believed was some unused code.  What we see with this report is that while `app_rpt` does not send any `!IAXKEY!` messages we may get them from other applications.

Knowing the source will help us confirm any fixes.

wd5m, 2025-07-01:
This is probably the old Iaxrpt Windows client ([https://wiki.allstarlink.org/wiki/Setup/Download_IAXRpt](wiki.allstarlink.org/wiki/Setup/Download_IAXRpt)). I just confirmed seeing this warning in my Asterisk messages log file. Apparently from my use of Iaxrpt client.

David, WD5M

Allan-N, 2025-07-01:
For reference : [IAX Text Protocol](https://wiki.allstarlink.org/wiki/IAX_Text_Protocol)

Allan-N, 2025-07-01:
Prior to v3.4.x, it looks like the code would check for (and essentially ignore) the `!IAXRPT!` message.  Now, currently, we are no longer ignoring the message.

Allan-N, 2025-07-01:
>Might not be related but I can not get outstreamcmd= to work

I would be surprised if the `!IAXKEY!` warning messages would have any impact on `outstreamcmd=`.

Q? have you looked at [Streaming Node Audio to Broadcastify](https://allstarlink.github.io/adv-topics/broadcastify/) ?

If that doesn't help, I would suggest first going to [AllStarLink Community](https://community.allstarlink.org) for help or, if warranted, create a new/separate issue here on GitHub.

Allan-N, 2025-07-02:
Re: the `!IAXKEY!` message.

I have succeeded in reproducing the warning message (that appears to be harmless).

... and I really dislike Windows
... and I really dislike Windows
... and I really dislike Windows

closed 2025-07-20 by Allan-N