AllStarLink/app_rpt #714: archivedir generates incomplete audio recordings under certain conditions

kenalker, 2025-07-02:
**Describe the bug**
I am using the “archivedir” function to record repeater audio.  This function occasionally produces incomplete WAV files.

**To Reproduce**
Steps to reproduce the behavior:
1) key up system via AllStar
2) double with AllStar session by keying up an FM transmitter
3) notice that recording of AllStar session terminates at start of FM session, and recording is incomplete
4) unkey FM transmitter and then unkey AllStar
5) notice that audio after unkey of FM transmitter is never recorded

My next post will have a much more detailed explanation.

**Expected behavior**
I expect to either have a complete WAV (*.wav) file recorded of the original AllStar session upon key up of FM transmitter, or I expect the recording to continue until session is complete (ie. all AllStar and FM transmissions are complete and FINAL key-down is detected).  Which method should occur is debatable, but second method is likely the most practical.  See my next post for more details.

**Software versions (listed in asl-menu, option 4)**
```
OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.1.0-32-amd64

Asterisk      : 22.2.0+asl3-3.3.0-6.deb12
ASL [app_rpt] : 3.3.0

Installed ASL packages :

Package                         Version
==============================  ==============================
allmon3                         1.4.2-1.deb12
asl3                            3.7.1-1.deb
asl3-asterisk                   2:22.2.0+asl3-3.3.0-6.deb12
asl3-asterisk-config            2:22.2.0+asl3-3.3.0-6.deb12
asl3-asterisk-dev               2:22.2.0+asl3-3.3.0-6.deb12
asl3-asterisk-doc               2:22.2.0+asl3-3.3.0-6.deb12
asl3-asterisk-modules           2:22.2.0+asl3-3.3.0-6.deb12
asl3-menu                       1.12-1.deb12
dahdi                           1:3.1.0-2
dahdi-dkms                      1:3.4.0-6+asl
dahdi-firmware-nonfree          2.11.1.0.20170917-2
dahdi-linux                     1:3.4.0-6+asl
dahdi-source                    1:3.4.0-6+asl
```

**Have you run a software update and rebooted?**
Yes.

**What is the platform - e.g. Raspberry Pi 4, Raspberry Pi 5, Virtual Machine, Desktop, etc.**
Beelink SER5 Mini PC with AMD Ryzen 5 5560U 6 core (up to 4.0GHz) 16GB DDR4 RAM, 500GB NVMe M.2 SSD, RX Vega7

Allan-N, 2025-07-02:
>During troubleshooting, it may help to follow the YYYYMMDD.txt file that is produced along side the WAV files. However, I was getting a bit confused by the contents of this file and started to suspect there may be some inconsistencies in the way the code is handling the logging in this file.

Here, there are two messages in the .txt file of interest.  We add the `CONNECT` message just after we have started capturing an audio archive.  You will then see the `DISCONNECT` message with the audio archive being stopped just after any associated telemetry (e.g. the courtesy tone) has completed.

kenalker, 2025-07-02:
I am not seeing those messages in my 20250702.txt file.  I just stripped, sorted, and uniq'd 20250702.txt and the complete list of keywords I am finding is as follows: "LINK, LINKD, RXKEY, RXUNK, TXKEY, TXUNK".  Perhaps I am misunderstanding your response.

Allan-N, 2025-07-03:
>I am not seeing those messages in my 20250702.txt file. I just stripped, sorted, and uniq'd 20250702.txt and the complete list of keywords I am finding is as follows: "LINK, LINKD, RXKEY, RXUNK, TXKEY, TXUNK". Perhaps I am misunderstanding your response.

My bad.  I was not looking at the code correctly (missed and mis-read a few spots).  The log messages indicating that we are starting an audio archive are `TXKEY,MAIN`, `RXKEY,MAIN`, and `CONNECT`.   The log messages indicating when we stop the audio archiving are `TXUNKEY,MAIN`, `RXUNKEY,MAIN`, and `DISCONNECT`.

mkmer, 2025-08-07:
This seems a bit odd all around.  We start/stop the recording here:
https://github.com/AllStarLink/app_rpt/blob/13d2b5332d798c416415745a21d0de1dabb1f058/apps/app_rpt.c#L5143-L5150

Based on the `totx` solve.  It looks like a "one shot" based on the transmitter keying/unkeying. 
I don't see (at the moment) how we could "toggle" `totx` when one or the other "unkeys" or "keys" as long as someone is "keyed" it should stay "true".  It MUST be toggling for the behavior to occur?!?

Maybe we "don't" want to close the stream IF it's currently recording and we trigger again here:
https://github.com/AllStarLink/app_rpt/blob/13d2b5332d798c416415745a21d0de1dabb1f058/apps/app_rpt.c#L2975-L2977
We could just leave the stream open, don't generate a new file name, and skip the `do_archive` logic.

We really should never get here with stream still open.