AllStarLink/app_rpt #734: app_rpt: Add Key-up Delay Configurable Option

ve7fet, 2025-07-29:
Many analog controllers have an option to delay the assertion of a valid COS/COR signal until that signal has been present for a configurable amount of time.

This feature is used to reduce "kerchunking" of the repeater due to quick bursts of noise, or users "bumping their PTT" (intentionally, or unintentionally).

This is **not** the same as timers/delays used for squelch tail elimination.

For example, if `keyupdelay` is set to `250`(mS), then COS/COR has to be valid for that minimum time, before the repeater will be keyed. Any squelch openings shorter than that will just be ignored.

Allan-N, 2025-07-30:
Would this delay force one to PTT, *pause*, and then speak?

And, yes, folks should be doing this anyway but I wonder if this might result in more voice truncation at the start of a transmission.

KB4MDD, 2025-07-30:
This is related to #90, Kerchunk filter.  I used this on my ACC and RLC controllers.

This is from the ACC RC96 manual.

> An algorithm fIlters against spurious noise on the receiver COS logic input, preventing static crashes and random noise bursts from bringing up the repeater. Kerchunker filtering prevents short transmissions from holding up the repeater transmitter, frustrating the kerchunker into making a "real" transmission to see if the repeater is up. The kerchunker filter requires that a user key down for more than 750 ms or an instant hang time is generated. 

The RLC3 manual has the keyup delay timer.

> The keyup delay timer is used to keep short noise bursts that open the receiver’s squelch from keying up the repeater.  It can also keep people from kerchunking the repeater.  When enabled, it  makes the controller totally ignore keyups shorter than the length of this timer (the rest of this explanation will assume ½ second).  If the receiver is keyed for longer than ½ second, the first ½ second will be ignored, then it will respond normally, bringing up the transmitter, triggering the ID system, etc.  To keep every keyup from being delayed, the keyup delay is automatically disabled while the repeater is being used.  It is only re-enabled when the repeater is not used for a default of 60 seconds.

mkmer, 2025-08-07:
Isn't this what the `rx_ondelay` does for us (in the channel driver)?