AllStarLink/app_rpt #743: chan_simpleusb.c: Config Variables With No Default Value

ve7fet, 2025-08-04:
As I am working on documenting `simpleusb.conf`, and going through all the available options/variables that can be set, it appears that the following do not have default values defined if they do not appear in `simpleusb.conf`:

```
invertptt
carrierfrom
ctcssfrom
rxboost
hdwtype
plfilter
deemphasis
preemphasis
duplex3
```

Allan-N, 2025-08-04:
It looks like each `struct chan_simpleusb_pvt` will be initialized in [store_config()](https://github.com/AllStarLink/app_rpt/blob/171c7f27a66276e106ad24bd0f07c59832ef4d4a/channels/chan_simpleusb.c#L3714) with the default values being pulled from [simpleusb_default](https://github.com/AllStarLink/app_rpt/blob/171c7f27a66276e106ad24bd0f07c59832ef4d4a/channels/chan_simpleusb.c#L334) or simply being zero'd out.  In both cases, any struct variable will, unless set otherwise, start off 0/NULL.

ve7fet, 2025-08-04:
Ok, that works. I will document the settings with defaults of 0.

closed 2025-08-04 by ve7fet