AllStarLink/app_rpt #745: chan_simpleusb.c: txoffdelay Not Capped to 400s

ve7fet, 2025-08-05:
In the Asterisk read function:

https://github.com/AllStarLink/app_rpt/blob/171c7f27a66276e106ad24bd0f07c59832ef4d4a/channels/chan_simpleusb.c#L2283-L2293

The above code purports to cap the `txoffdelay` (via  `txoffcnt`) at 20000 (400s). However, that is not the case, since it is comparing it to see if `txoffcnt` has reached 50000 (1000s) before capping the timer to 20000.

Allan-N, 2025-08-05:
FYI: the same chunk of code also exists in chan_usbradio.c

mkmer, 2025-08-05:
While the documentation is wrong (and it's odd), I don't think it's broken:
https://github.com/AllStarLink/app_rpt/blob/171c7f27a66276e106ad24bd0f07c59832ef4d4a/channels/chan_simpleusb.c#L2296.

The above code just keeps the counter from rolling past 50k, (aka our test should not be > 20k in ln2296)
Definitely something to "fix', but it looks like it's doing what it was intended to do

On the other hand: It doesn't appear that we limit values in `o->txoffdelay` - so it's possible to never get there, OR be between 20k and 50k -> that will be weird..
https://github.com/AllStarLink/app_rpt/blob/171c7f27a66276e106ad24bd0f07c59832ef4d4a/channels/chan_simpleusb.c#L3558-L3561

ve7fet, 2025-08-05:
Yes, it looks like the intent was to cap it at 20000, so you'd figure line 2289 should be `if (o->txoffcnt > 20000) {`. Otherwise, you have this weird situation where if someone for whatever reason set `txoffdelay` to `50001`, it would time out after 20000. 

Not really a critical bug... more of a code quality issue. 

Even 20000 (400s) seems like an unreasonably long time to want to have the transmitter be off before the node is allowed to receive again. Almost like someone got confused and thought the timer was in ms (ie 20s), instead of x * 20ms. Or is it even time at all? It is actually just a counter, so there isn't really a time attached to it, is there? So it would actually be multiples of the time it takes to service the function?

PR #748 (2025-08-08): chan_usbradio & chan_simpleusb: add limits.
chan_simpleusb:
address incorrect limit on o->txoffcnt.
Add limits to txoffdelay and rxondelay.

Fixes #745 
Seems like 1 minute on/off delay is plenty of time for this configuration.
closed 2025-08-08 by mkmer