AllStarLink/app_rpt #753: chan_usbradio: Enabling Parallel Port Pins Causes Asterisk to Hang on Restart

ve7fet, 2025-08-09:
The code to manage the parallel port pins between `chan_simpleusb` and `chan_usbradio` is nearly the same (there are a few differences). 

When parallel port pins are enabled on `chan_simpleusb`, everything appears to operate normally.

Doing the same thing on `chan_usbradio` causes Asterisk to hang when doing a `sudo systemctl restart asterisk.service` to pick up the changes in `usbradio.conf`, and the parallel port pin functions do not work.

Steps to reproduce:

* find yourself a PC with a parallel port (yeah, I know...)
* configure your USB radio interface with `chan_simpleusb`
* add `pp12 = in` to `simpleusb.conf`
* restart Asterisk with `sudo systemctl restart asterisk.service`
* from the Asterisk CLI, verify `pp12` status shows up with `rpt show variables <nodenumber>`
* reconfigure for `chan_usbradio`
* add `pp12 = in` to `usbradio.conf`
* restart Asterisk with `sudo systemctl restart asterisk.service`
* observe the process hang

Once Asterisk gets hung up, it's stuck hard. Have to kill Asterisk manually, and systemctl times out restarting it (but it does seem to restart, but then the canary dies).

Did find one message in the Asterisk logs:

`[2025-08-08 17:25:02.501] ERROR[1027] res_usbradio.c: Unable to claim printer port /dev/parport0, disabling pp support`

But, it is odd that `chan_simpleusb` has no similar problems.

A possible related issue is that `ast_radio_load_parallel_port` in `res_usbradio.c` claims the parallel port (PPCLAIM), but it doesn't look like it cleans up after itself with a PPRELEASE when it closes the ppfd.

mkmer, 2025-08-13:
Maybe we are not handling said failure gracefully? Or a typo in chan_usbradio on how we handle it???
`res_usbradio.c: Unable to claim printer port /dev/parport0, disabling pp support`

ve7fet, 2025-08-13:
I'm working on spinning up a dev box so I can install ASL3 compiled with the debug options, so I can see if I can gather more data.

`chan_simpleusb` operates correctly. It is `chan_usbradio` that fails. The only difference I see between the functions that the channel driver uses is that `chan_usbradio` makes some calls to `xpmr` that `chan_simpleusb` does not.

ve7fet, 2025-08-14:
Well, the plot thickens.

Cold boot, no "pp" pins enabled, everything is fine.

```
asl@asl-testbed:/etc/asterisk$ sudo tail usbradio.conf
devstr =
rxmixerset = 500
txmixaset = 500
txmixbset = 500
rxvoiceadj = 0.500000
rxctcssadj = 0.500000
txctcssadj = 200
rxsquelchadj = 500
;pp12 = in
fever = 0
```

```
asl@asl-testbed:/etc/asterisk$ sudo asl-find-sound
7-2:1.0 -->     0d8c:000c
```

I can access `radio-tune-menu`, and monitor the status of the on-board GPIOs (`v`).

Edit `usbradio.conf` to un-comment `pp12 = in`.

```
CLI:
asl-testbed*CLI> module refresh chan_usbradio
Unable to unload resource chan_usbradio
Command 'module refresh chan_usbradio' failed.
[2025-08-14 07:14:23.165] WARNING[723]: loader.c:1370 auto_unload_resource: Soft unload failed, 'chan_usbradio' has use count 1
```

```
Log:
[2025-08-14 07:14:23.165] WARNING[723] loader.c: Soft unload failed, 'chan_usbradio' has use count 1
```

Check the variables to see if PP12 shows up:

```
CLI:
asl-testbed*CLI> rpt show variables 1999
Variable listing for node 1999:
   RPT_ALINKS=0
   RPT_LINKS=0
   RPT_NUMALINKS=0
   RPT_NUMLINKS=0
   RPT_AUTOPATCHUP=0
   RPT_ETXKEYED=0
   RPT_TXKEYED=0
   RPT_RXKEYED=0
    -- 8 variables
```

Nope.

Do a `module reload` in the CLI:

```
CLI:
asl-testbed*CLI> module reload
    -- Reloading module 'extconfig' (Configuration)
    -- Reloading module 'logger' (Logger)
 Asterisk Queue Logger restarted
    -- Reloading module 'cdr' (CDR Engine)
[2025-08-14 07:16:25.913] NOTICE[723]: cdr.c:4639 cdr_toggle_runtime_options: CDR simple logging enabled.
    -- Reloading module 'cel' (CEL Engine)
[... 306 lines cut ...]
[2025-08-14 07:39:09.582] ERROR[1383] chan_usbradio.c: Channel 1999: Device string 7-2:1.0 is already assigned to channel 1999
[2025-08-14 07:39:10.193] ERROR[1383] chan_usbradio.c: Channel 1999: Device string 7-2:1.0 is already assigned to channel 1999
[2025-08-14 07:39:10.806] ERROR[1383] chan_usbradio.c: Channel 1999: Device string 7-2:1.0 is already assigned to channel 1999
[2025-08-14 07:39:11.418] ERROR[1383] chan_usbradio.c: Channel 1999: Device string 7-2:1.0 is already assigned to channel 1999
[2025-08-14 07:39:12.030] ERROR[1383] chan_usbradio.c: Channel 1999: Device string 7-2:1.0 is already assigned to channel 1999
```

The CLI response appears to be response for the `module reload` at 07:24 as well as 07:39.

The `ERROR[1383] chan_usbradio.c: Channel 1999: Device string 7-2:1.0 is already assigned to channel 1999` now repeats constantly. The device string 7-2:1.0 indeed is supposed to be assigned to 1999. I suppose it re-read `usbadio.conf` and is now all confused (even though it is the correct `devstr`).

I do see in the CLI/Log:

```
[2025-08-14 07:39:07.469] ERROR[574]: res_usbradio.c:708 ast_radio_ppread: Unable to read pp dev /dev/parport0
[2025-08-14 07:39:07.469] ERROR[723]: res_usbradio.c:672 ast_radio_load_parallel_port: Unable to claim printer port /dev/parport0, disabling pp support
```

So, was it hung up trying to talk to/claim the parallel port?

[core-asterisk-running-2025-08-14T14-38-46Z-brief.txt](https://github.com/user-attachments/files/21775422/core-asterisk-running-2025-08-14T14-38-46Z-brief.txt)
[core-asterisk-running-2025-08-14T14-38-46Z-full.txt](https://github.com/user-attachments/files/21775420/core-asterisk-running-2025-08-14T14-38-46Z-full.txt)
[core-asterisk-running-2025-08-14T14-38-46Z-info.txt](https://github.com/user-attachments/files/21775418/core-asterisk-running-2025-08-14T14-38-46Z-info.txt)
[core-asterisk-running-2025-08-14T14-38-46Z-locks.txt](https://github.com/user-attachments/files/21775421/core-asterisk-running-2025-08-14T14-38-46Z-locks.txt)
[core-asterisk-running-2025-08-14T14-38-46Z-thread1.txt](https://github.com/user-attachments/files/21775419/core-asterisk-running-2025-08-14T14-38-46Z-thread1.txt)