AllStarLink/app_rpt #754: Time out timer functionallity broken

hwstar, 2025-08-09:
I found a bug in the the master branch of app_rpt commit: 13d2b53
I was chatting with someone after a lengthy response who caused a time out on my end. When I keyed my receiver the time out was reset, but then afterwards my node TX would not hold up during the time the far end was talking.

Steps to repeat:

1. Change your config file to shorten the time out timer so that you can see some time outs.
2. Connect to a net known to have users who talk for a long time
3. Wait for your node to time out
4. Reset the time out timer by supplying a signal on the input frequency of your node (Far end must be actively talking to reproduce the bug)
5. You will hear talking, but then your node will stop transmitting (The far end no longer holds the transmitter up)
6. Keying up again will cause the same behavior to repeat 
7. If the far end drops and becomes inactive, the next time someone uses it, the operation will be normal until the next time out.

I will start checking out prior versions of app_rpt.c  from git to see which commit introduced the problem. I will post those findings here once they're known.

hwstar, 2025-08-09:
The issue lies in the differences between 13d2b53 and 24a0f2f. This update was supposed to help mitigate picket fencing resetting the time out timer, but it looks like it broke a feature from way back when: To reset the time out timer when some one long winded is talking on the far side of a connection one could key up on the near end receiver. This would result in the time out timer getting reset so that the far end station could be heard again. 

So the time out timer logic should operate in two modes. One where there is picket fencing and there is a picket fencing signal on the near end receiver, and the other when a user is listening to a far end QSO and the time out timer times out. In the latter case, the logic should be to reset the timer if a signal is detected on the input for a small amount of time (maybe the same value used for the picket fencing timer).

hwstar, 2025-08-10:
The problem is due to a change on line 5011 which added an additional logical and statement for myrpt->toresettimer so that the time out timer could be not be reset without waiting for the time out reset timer to expire.

The problem can to be resolved if myrpt->remrx is or'd with  myrpt->remrx which will allow the time out timer to be reset when there is an active station transmitting on the far end.  Here is the modified line:

```
	/* If unkey and re-key, reset time out timer */
		if ((!totx) && (!myrpt->totimer) && (!myrpt->tounkeyed) && (!myrpt->keyed) && ((!myrpt->toresettimer) || myrpt->remrx)) {
			myrpt->tounkeyed = 1;
		}
```

Now historically,  when the far end station's transmission caused a timeout, you could reset  the timeout by doing a brief key and unkey on the local receiver. The enhancement to add the time out timer reset timer changed functionality. Adding myrpt-remrx as an override to to the time out timer reset timer is one way to address this, but it has made me think that that original time out timer reset feature could be flawed when used in conjunction with with the new time out timer reset feature.  Here's where it falls down: If there is a lot of noise opening and closing the ctcss and/or squelch on the local receiver and no one is around, the signal on the far end tx could cause the timeout period to be ignored and the local tx will stay keyed with the remote signal until it drops.

Still, if nothing is done, then  it is still a bug (and a rather confusing one at that). If the time out timer reset timer feature is enabled by setting the it to a non-zero value, then any local user who tries to reset it by sending a brief transmission to the local receiver will cause a repeater tail to be heard  along with the far end audio, but once the tail timer expires, the tx drops right in the middle of the QSO. This still has to be fixed regardless.

mkmer, 2025-08-11:
Probably should put this functionality in the manual -> currently that is not how it is "described" :(
https://allstarlink.github.io/config/rpt_conf/#totime

Allan-N, 2025-08-11:
It would be nice to capture the essence of the above comments IN THE CODE.  I would also be in favor of making this part of the code more readable.

hwstar, 2025-08-11:
@allan: Agree 100%

mkmer, 2025-08-13:
@hwstar  - IF you have some time, could you put the fix + proposed timer in a PR -> it sounds good (to me).

mkmer, 2025-08-13:
Thanks!  I too am hardwareless at the moment (On "vacation").  I will test it best I can when I get back to some hardware :)

Allan-N, 2025-08-13:
Just thinking about the variables associated with the timeout reset and burping and I'm wondering how we can make things a bit less cryptic/tech-y/geek-y (both in the .conf files AND in the code).

We recently added (but have not yet released) `toresettimer` but maybe a more descriptive name might be `timeout_reset_keydown_time`?

Now, we're talking about adding `toresetburp` but maybe a more descriptive name might be `timeout_reset_keyup_time`? or `timeout_reset_kerchunk_time`?

hwstar, 2025-08-14:
The pull request for these fixes and enhancements is #765

hwstar, 2025-08-15:
@mkmer Need you to re-confirm something. Testing has revealed that the time out timer gets reset just after the courtesy tone is sent, and it seems to be debounced so that in and out signals between the unkey and the courtesy tone will not reset time out timer. 

Your fix in #744  will only take effect once a time out has occurred and the TX has been disabled. 

We already have filtering before the CT to prevent the TOT from getting reset at that point in the code. So you want the RX to not reset the tot for some interval after the tot expires and the TX is disabled, correct? 

Just want to make sure we are on the same page.

hwstar, 2025-08-25:
PR770 Merged.

closed 2025-08-25 by hwstar