AllStarLink/app_rpt #763: channel.c:1105 __ast_queue_frame: Exceptionally long queue length

ve7fet, 2025-08-14:
So, my ASL3 went off the rails tonight.

This is ASL3 (versions below), running on a Pi 4. It has three nodes, two are `chan_voter` (435451/435452) and one is `chan_simpleusb` (1999).

Received an email from a member that the repeater had been stuck on transmit. "for a very long time". I checked Allmon3, and it appeared to not be able to connect to the manager, as nothing was being shown for the nodes. 

Logged in via SSH, Asterisk was running. Connected to the console, and started seeing warnings from `channel.c`:

```
Connected to Asterisk 22.4.1+asl3-3.5.4-1.deb12 currently running on ve7rpt (pid = 2001)
[2025-08-13 22:18:32.866] WARNING[2033]: channel.c:1105 __ast_queue_frame: Exceptionally long queue length (0 voice / 398 total) queuing to IAX2/44.32.40.212:4569-4215
[2025-08-13 22:18:42.866] WARNING[2040]: channel.c:1105 __ast_queue_frame: Exceptionally long queue length (0 voice / 399 total) queuing to IAX2/44.32.40.212:4569-4215
[2025-08-13 22:18:52.866] WARNING[2039]: channel.c:1105 __ast_queue_frame: Exceptionally long queue length (0 voice / 400 total) queuing to IAX2/44.32.40.212:4569-4215
```

Did a `module reload` from the CLI, and got dumped back to my shell prompt. Reconnected to the Asterisk CLI to catch some of the messages:

```
Connected to Asterisk 22.4.1+asl3-3.5.4-1.deb12 currently running on ve7rpt (pid = 2001)
[2025-08-13 22:18:32.866] WARNING[2033]: channel.c:1105 __ast_queue_frame: Exceptionally long queue length (0 voice / 398 total) queuing to IAX2/44.32.40.212:4569-4215
[2025-08-13 22:18:42.866] WARNING[2040]: channel.c:1105 __ast_queue_frame: Exceptionally long queue length (0 voice / 399 total) queuing to IAX2/44.32.40.212:4569-4215
[2025-08-13 22:18:52.866] WARNING[2039]: channel.c:1105 __ast_queue_frame: Exceptionally long queue length (0 voice / 400 total) queuing to IAX2/44.32.40.212:4569-4215
```

Repeater stopped transmitting, and normal operations appeared to have resumed.

Start of event appears to be `[2025-08-13 21:27:04.109]`. I did the `module reload` apparently at `Aug 13 22:20:59 `.

Versions:

```
OS            : Debian GNU/Linux 12 (bookworm)
OS Kernel     : 6.12.34+rpt-rpi-v8
Asterisk      : 22.4.1+asl3-3.5.4-1.deb12
ASL [app_rpt] : 3.5.4
Installed ASL packages :
Package                         Version
==============================  ==============================
allmon3                         1.5.1-1.deb12
asl3                            3.8-1.deb
asl3-asterisk                   2:22.4.1+asl3-3.5.4-1.deb12
asl3-asterisk-config            2:22.4.1+asl3-3.5.4-1.deb12
asl3-asterisk-modules           2:22.4.1+asl3-3.5.4-1.deb12
asl3-menu                       1.14-1.deb12
```

Attached are the Asterisk log and journalctl log from the event period. No debugging symbols in the installed packages, so it looks like I can't get a backtrace.

[messages.log](https://github.com/user-attachments/files/21766841/messages.log)
[journalctl.log](https://github.com/user-attachments/files/21766840/journalctl.log)

ve7fet, 2025-09-03:
Got hit by this again on the same node during a net this morning.

However, this time it appears to have been sending to `chan_voter`:

```
[2025-09-03 11:01:31.422] WARNING[1491848] channel.c: Exceptionally long queue length (3 voice / 131 total) queuing to voter/435452
[2025-09-03 11:01:31.442] WARNING[1491848] channel.c: Exceptionally long queue length (3 voice / 131 total) queuing to voter/435452
```

Restarting Asterisk took forever, but I think it mostly recovered... ended up just rebooting to start clean (was just trying to get it back on the air quickly).

I think Asterisk had mostly stopped responding, as the Allmon3 on our Hub (hosted on a VPS) couldn't talk to the node anymore.

InterLinked1, 2025-09-03:
> Got hit by this again on the same node during a net this morning.
>
> However, this time it appears to have been sending to `chan_voter`:
>
> ```
> [2025-09-03 11:01:31.422] WARNING[1491848] channel.c: [...]

Did you get a backtrace by chance? Without knowing what was going on, it's difficult if not impossible to investigate.

mkmer, 2025-09-04:
This "seems" a bit odd to me:
```
Thread 5 (Thread 0x7f6d5bb040 (LWP 2049)):
#0  determine_info (symbolp=0x0, mapp=0x7f6d5b7d30, info=0x7f6d5b7c30, match=0x7fbd422380, addr=366795177908) at ./elf/dl-addr.c:49
#1  _dl_addr (address=0x5566b23fb4 <ast_log_backtrace+36>, info=info@entry=0x7f6d5b7c30, mapp=mapp@entry=0x7f6d5b7d30, symbolp=symbolp@entry=0x0) at ./elf/dl-addr.c:131
#2  0x0000007fbc289838 in __backtrace_symbols (array=array@entry=0x7f70001740, size=size@entry=9) at ./debug/backtracesyms.c:48
#3  0x00000055669714c0 in __ast_bt_get_symbols (addresses=addresses@entry=0x7f70001740, num_frames=9) at ./main/backtrace.c:296
#4  0x0000005566b23fcc in ast_log_backtrace () at ./main/logger.c:2497
#5  0x0000005566ad0230 in __ast_assert_failed (condition=condition@entry=0, condition_str=condition_str@entry=0x5566b57bb8 "user_data is NULL", file=file@entry=0x5566b60f98 "channel.c", line=line@entry=2503, function=function@entry=0x5566b65ca8 <__PRETTY_FUNCTION__.187> "ast_softhangup") at ./main/utils.c:2828
#6  0x0000005566963c18 in log_bad_ao2 (func=0x5566b65ca8 <__PRETTY_FUNCTION__.187> "ast_softhangup", line=2503, file=0x5566b60f98 "channel.c", user_data=0x0) at ./main/astobj2.c:212
#7  __ao2_lock (user_data=user_data@entry=0x0, lock_how=lock_how@entry=AO2_LOCK_REQ_MUTEX, file=file@entry=0x5566b60f98 "channel.c", func=func@entry=0x5566b65ca8 <__PRETTY_FUNCTION__.187> "ast_softhangup", line=line@entry=2503, var=var@entry=0x5566b5e558 "chan") at ./main/astobj2.c:224
#8  0x00000055669a57b0 in ast_softhangup (chan=0x0, cause=cause@entry=1) at ./main/channel.c:2503
#9  0x0000007fb8a6c6dc in rpt (this=this@entry=0x7fb8ad20d0 <rpt_vars>) at ./apps/app_rpt.c:4910
#10 0x0000005566acab0c in dummy_start (data=<optimized out>) at ./main/utils.c:1607
#11 0x0000007fbc212030 in start_thread (arg=0x0) at ./nptl/pthread_create.c:442
#12 0x0000007fbc27bf5c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:79
```
This appears to indicate we are trying to reload the repeater???

https://github.com/AllStarLink/app_rpt/blob/2c721d13d13fcc39b10123eac9427d72d5957564/apps/app_rpt.c#L4910

Not sure I'm reading this correctly.

mkmer, 2025-09-04:
@InterLinked1 : What does this mean? 
```Exceptionally long queue length (0 voice / 400 total)``` 
-> the channel queue has 400 frames but none of them are voice frames?

jxmx, 2025-09-24:
I have started to encounter this frequently. Do we have any debugging ideas?

jxmx, 2025-09-24:
A few more specific details… This is occurring on  a Pi4 that is fed by a very busy hub now that is running in the cloud. I do not see a corresponding event in the hub node that matches when this goes off the rails. However, I have also noticed that there appears to be one or two nodes trying to connect directly, which is unusual, but it shouldn’t cause a problem. The only thing I can think of is if there is a momentary the Internet would that throw this into a loop?

InterLinked1, 2025-09-24:
Does your log message look the same? I wonder if it would help to add a temporary debugging log message to show what type of frames are being queued up, maybe that might provide a clue?

jxmx, 2025-09-25:
So in looking through this, there is no surrounding context to the error. The only interesting thing is there's someone sending DTMF tones each time about 25 minutes before:

```
[2025-09-22 06:38:09.118] DTMF[1015827] channel.c: DTMF begin '8' received on SimpleUSB/43211
[2025-09-22 06:38:09.118] DTMF[1015827] channel.c: DTMF begin passthrough '8' on SimpleUSB/43211
[2025-09-22 06:38:09.224] DTMF[1015827] channel.c: DTMF end '8' received on SimpleUSB/43211, duration 106 ms
[2025-09-22 06:38:09.224] DTMF[1015827] channel.c: DTMF end accepted with begin '8' on SimpleUSB/43211
[2025-09-22 06:38:09.224] DTMF[1015827] channel.c: DTMF end passthrough '8' on SimpleUSB/43211
[2025-09-22 06:38:09.325] DTMF[1015827] channel.c: DTMF begin '0' received on SimpleUSB/43211
[2025-09-22 06:38:09.325] DTMF[1015827] channel.c: DTMF begin passthrough '0' on SimpleUSB/43211
[2025-09-22 06:38:09.439] DTMF[1015827] channel.c: DTMF end '0' received on SimpleUSB/43211, duration 113 ms
[2025-09-22 06:38:09.439] DTMF[1015827] channel.c: DTMF end accepted with begin '0' on SimpleUSB/43211
[2025-09-22 06:38:09.439] DTMF[1015827] channel.c: DTMF end passthrough '0' on SimpleUSB/43211
[2025-09-22 06:38:09.539] DTMF[1015827] channel.c: DTMF begin '8' received on SimpleUSB/43211
[2025-09-22 06:38:09.539] DTMF[1015827] channel.c: DTMF begin passthrough '8' on SimpleUSB/43211
[2025-09-22 06:38:09.672] DTMF[1015827] channel.c: DTMF end '8' received on SimpleUSB/43211, duration 133 ms
[2025-09-22 06:38:09.672] DTMF[1015827] channel.c: DTMF end accepted with begin '8' on SimpleUSB/43211
[2025-09-22 06:38:09.672] DTMF[1015827] channel.c: DTMF end passthrough '8' on SimpleUSB/43211
[2025-09-22 06:38:09.758] DTMF[1015827] channel.c: DTMF begin '0' received on SimpleUSB/43211
[2025-09-22 06:38:09.758] DTMF[1015827] channel.c: DTMF begin passthrough '0' on SimpleUSB/43211
[2025-09-22 06:38:09.886] DTMF[1015827] channel.c: DTMF end '0' received on SimpleUSB/43211, duration 128 ms
[2025-09-22 06:38:09.886] DTMF[1015827] channel.c: DTMF end accepted with begin '0' on SimpleUSB/43211
[2025-09-22 06:38:09.886] DTMF[1015827] channel.c: DTMF end passthrough '0' on SimpleUSB/43211
[2025-09-22 06:38:09.987] DTMF[1015827] channel.c: DTMF begin '2' received on SimpleUSB/43211
[2025-09-22 06:38:09.987] DTMF[1015827] channel.c: DTMF begin passthrough '2' on SimpleUSB/43211
[2025-09-22 06:38:10.099] DTMF[1015827] channel.c: DTMF end '2' received on SimpleUSB/43211, duration 112 ms
[2025-09-22 06:38:10.099] DTMF[1015827] channel.c: DTMF end accepted with begin '2' on SimpleUSB/43211
[2025-09-22 06:38:10.099] DTMF[1015827] channel.c: DTMF end passthrough '2' on SimpleUSB/43211
```

And then poof:

```
[2025-09-22 07:07:47.217] WARNING[1015814] channel.c: Exceptionally long voice queue length (97 voice / 98 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:48.497] WARNING[1015818] channel.c: Exceptionally long voice queue length (97 voice / 98 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:49.777] WARNING[1015813] channel.c: Exceptionally long voice queue length (97 voice / 98 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:51.057] WARNING[1015813] channel.c: Exceptionally long voice queue length (97 voice / 98 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:52.337] WARNING[1015815] channel.c: Exceptionally long voice queue length (97 voice / 98 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:54.602] WARNING[1015814] channel.c: Exceptionally long voice queue length (97 voice / 100 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:55.882] WARNING[1015820] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:57.162] WARNING[1015814] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:58.442] WARNING[1015814] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:07:59.722] WARNING[1015813] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:08:01.002] WARNING[1015819] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:08:02.282] WARNING[1015816] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:08:03.562] WARNING[1015820] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:08:04.842] WARNING[1015814] channel.c: Exceptionally long voice queue length (97 voice / 101 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:08:06.122] WARNING[1015822] channel.c: Exceptionally long voice queue length (97 voice / 102 total) queuing to IAX2/3.138.188.232:4569-6613
[2025-09-22 07:08:07.402] WARNING[1015822] channel.c: Exceptionally long voice queue length (97 voice / 102 total) queuing to IAX2/3.138.188.232:4569-6613
```

This spools ad nauseum until the system is restarted. I had originally thought this was related to some broken node connecting to our system but nothing was connected during this error except 3.138.188.232 which is our hub in AWS. It did not have any logged problems. The ONLY item that appeared was two minutes before:

```
[2025-09-22 07:05:14.789] WARNING[91411][C-000035fe] func_channel.c: Unknown or unavailable item requested: 'peerip'
[2025-09-22 07:05:14.789] WARNING[91411][C-000035fe] func_channel.c: Unknown or unavailable item requested: 'peerip'
```

I don't know what that means but seems unrelated.

InterLinked1, 2025-09-25:
> The only interesting thing is there's someone sending DTMF tones each time about 25 minutes before:

If it's that long beforehand, I doubt it's related.

> I don't know what that means but seems unrelated.

It's probably a dialplan function call using parameters that are invalid for the current channel technology type. There should be some dialplan associated with it, unless the console verbosity was less than 3 for some reason.

mkmer, 2025-09-26:
There was a mention back in March that also looks similar.
https://community.allstarlink.org/t/var-log-asterisk-messages-log-filling-up-taking-up-my-space/23028.

mkmer, 2025-09-27:
On more "thought" - Are all of the "problem" installations on Trixie?

jxmx, 2025-09-27:
No, in fact, they are on bookworm

jxmx, 2025-09-29:
This requires the debug symbols packages installed right?

mkmer, 2025-09-29:
> This requires the debug symbols packages installed right?

Not necessarily, it would be more helpful if they are included, but we can get "something" without it (I think anyway).
I'm far from an expert with these tools.

mkmer, 2025-10-15:
> Yeah, this is running on a Pi Appliance, so using the pre-compiled packages for ease of deployment.
> 
> However, `asl3-asterisk-dbgsym` and `asl3-asterisk-modules-dbgsym` ARE installed.

Do you have the `<coredump>-full.txt`, `<coredump>-info.txt`, `<coredump>-locks.txt` files? 
Or try to generate them using asl_coredumper from the coredump you have...

mkmer, 2025-10-15:
Well - I was hit by one today.  Core dump attached:

[core-asterisk-running-2025-10-15T18-47-57Z-brief.txt](https://github.com/user-attachments/files/22934753/core-asterisk-running-2025-10-15T18-47-57Z-brief.txt)
[core-asterisk-running-2025-10-15T18-47-57Z-full.txt](https://github.com/user-attachments/files/22934755/core-asterisk-running-2025-10-15T18-47-57Z-full.txt)
[core-asterisk-running-2025-10-15T18-47-57Z-info.txt](https://github.com/user-attachments/files/22934757/core-asterisk-running-2025-10-15T18-47-57Z-info.txt)
[core-asterisk-running-2025-10-15T18-47-57Z-locks.txt](https://github.com/user-attachments/files/22934758/core-asterisk-running-2025-10-15T18-47-57Z-locks.txt)
[core-asterisk-running-2025-10-15T18-47-57Z-thread1.txt](https://github.com/user-attachments/files/22934756/core-asterisk-running-2025-10-15T18-47-57Z-thread1.txt)

Also had our "new" warning message:
`[2025-10-15 14:45:47.871] WARNING[963098] app_rpt.c: RPT thread on 29972 is hung for 32 seconds.`

mkmer, 2025-10-16:
> Well - I was hit by one today. Core dump attached: [...]

I have to mark this as probably not related although it had the same "result".  With the difficulties I was having installing packages today I can't be positive which version of the app I was running when this crash occurred. 
One of my dev version had a deadlock and very possibly was the version that hung up here.

jxmx, 2025-10-17:
It's happened again for me. Here's the output and I kept the core too. With debug even cranked up to 9, all I get is:

```
[2025-10-17 12:25:31.812] WARNING[190827]: chan_iax2.c:3645 __attempt_transmit: Max retries exceeded to host 3.138.188.232 on IAX2/3.138.188.232:4569-7876 (type = 6, subclass = 11, ts=7571154, seqno=177)
[2025-10-17 12:25:35.814] WARNING[190835]: chan_iax2.c:3645 __attempt_transmit: Max retries exceeded to host 3.138.188.232 on IAX2/3.138.188.232:4569-7876 (type = 6, subclass = 2, ts=7575157, seqno=178)
[2025-10-17 12:25:41.812] WARNING[190828]: chan_iax2.c:3645 __attempt_transmit: Max retries exceeded to host 3.138.188.232 on IAX2/3.138.188.232:4569-7876 (type = 6, subclass = 11, ts=7581154, seqno=179)
```

The output of `core show locks` shows nothing:

```
node43211*CLI> core show locks
node43211*CLI>
```

[core-asterisk-running-2025-10-17T16-18-37Z-info.txt](https://github.com/user-attachments/files/22975732/core-asterisk-running-2025-10-17T16-18-37Z-info.txt)
[core-asterisk-running-2025-10-17T16-18-37Z-full.txt](https://github.com/user-attachments/files/22975729/core-asterisk-running-2025-10-17T16-18-37Z-full.txt)
[core-asterisk-running-2025-10-17T16-18-37Z-locks.txt](https://github.com/user-attachments/files/22975730/core-asterisk-running-2025-10-17T16-18-37Z-locks.txt)
[core-asterisk-running-2025-10-17T16-18-37Z-brief.txt](https://github.com/user-attachments/files/22975731/core-asterisk-running-2025-10-17T16-18-37Z-brief.txt)
[core-asterisk-running-2025-10-17T16-18-37Z-thread1.txt](https://github.com/user-attachments/files/22975733/core-asterisk-running-2025-10-17T16-18-37Z-thread1.txt)

InterLinked1, 2025-10-17:
> It's happened again for me. Here's the output and I kept the core too. With debug even cranked up to 9, all I get is:
>
> ```
> [2025-10-17 12:25:31.812] WARNING[190827]: chan_iax2.c:3645 [...]

Looking at the backtrace, there are quite a few threads waiting on locks, so it's strange that `core show locks` isn't showing any - might be related to how it was compiled.

I don't see anything related to the `chan_dahdi` interface lock though.

Allan-N, 2025-10-17:
One bit that I found in `core-asterisk-running-2025-10-17T16-18-37Z-full.txt` is that we have the manager thread exec'ing : `rpt cmd 43211 ilink 3 48496`

ve7fet, 2025-12-03:
Crashed again this morning.

okcrum, 2026-03-24:
Are we still seeing this with 3.8.2beta?

ve7fet, 2026-03-25:
I am still seeing it on 3.7.1 (had to reboot a node in the last couple of weeks because of it). When 3.8 gets pushed to mainline, I will upgrade and see what happens. It usually happens about once a month.